platformio/platform-ststm32 #463: Add SPL support for ST Nucleo F411RE

dg1167150, 2020-12-30:
What would it take to add Standard Peripheral Library framework to the ST Nucleo F411RE board?  I'm new to Platform IO and microcontrollers in general and can't seem to find the missing piece to make this work.  I've looked at the json file for the F401 board as an example and added 'spl' to the frameworks section of the nucleo_f411re.json file but there has to be something else as I am now getting a **KeyError: "Invalid board option 'build.core'"** from the build.

valeros, 2020-12-30:
Hi @dg1167150 ! The SPL framework is superseded by STM32Cube. We have no plans on updating support for SPL, only maintenance for legacy projects.

closed 2020-12-30 by valeros

ricehornet, 2021-02-09:
This tripped me up too today - @valeros I have opened a PR to update the docs to make this more clear.