platformio/platform-ststm32 #468: Linking Weak Symbols in STM32Cube framework code

a2k-hanlon, 2021-01-05:
I have a PlatformIO project using the STM32Cube framework. This framework uses weak symbols, and upon debugging the project, I can see that the weak functions have not been overridden. I believe I may be encountering the same issue as #205, where weak symbols are not being overridden when the linker is run due to the framework code being archived. Adding the configuration option `lib_archive=false` does not seem to resolve the issue. Looks like #205 was resolved, but the issue I'm encountering seems to be similar, albeit with a different library. The Arduino library was used in #205, whereas I am using STM32Cube.

This is the link command being executed, if it's any help:
```
arm-none-eabi-gcc -o .pio\build\robotdyn_blackpill_f303cc\firmware.elf -T C:\Users\Andrew\.platformio\packages\tool-ldscripts-ststm32\stm32f3\STM32F303CCTX_FLASH.ld -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 --specs=nano.specs --specs=nosys.specs .pio\build\robotdyn_blackpill_f303cc\src\main.o .pio\build\robotdyn_blackpill_f303cc\src\stm32f3xx_hal_msp.o .pio\build\robotdyn_blackpill_f303cc\src\stm32f3xx_it.o .pio\build\robotdyn_blackpill_f303cc\src\system_stm32f3xx.o -LC:\Users\Andrew\.platformio\platforms\ststm32\ldscripts -L.pio\build\robotdyn_blackpill_f303cc -LC:\Users\Andrew\.platformio\packages\framework-stm32cubef3\Drivers\CMSIS\Lib\GCC -LC:\Users\Andrew\.platformio\packages\framework-stm32cubef3\platformio\ldscripts -Wl,--start-group -lc -lgcc -lm -lstdc++ -lnosys .pio\build\robotdyn_blackpill_f303cc\libFrameworkHALDriver.a .pio\build\robotdyn_blackpill_f303cc\libFrameworkCMSISDevice.a -Wl,--end-group
```

a2k-hanlon, 2021-01-05:
Just adding for reference: the issue in #205 was initially brought forth here -> https://community.platformio.org/t/linking-libraries-with-weak-symbols/6302

valeros, 2021-01-05:
Hi @a2k-hanlon ! Could you please provide more info? What symbols wasn't overridden exactly? Where were they located?

Maybe you could prepare a minimal example?

a2k-hanlon, 2021-01-05:
Hi @valeros, thanks for such a quick response!

In my case, the symbols that are not being overwritten are `HAL_DAC_ConfigChannel()`, `HAL_DAC_Start()` and `HAL_DAC_SetValue()`. All 3 functions are declared in `stm32f3xx_hal_dac.h`, weakly defined in `stm32f3xx_hal_dac.c` and strongly defined in `stm32f3xx_hal_dac_ex.c`. These are files in the STM32Cube framework residing in
`<userfolder>\.platformio\packages\framework-stm32cubef3\Drivers\STM32F3xx_HAL_Driver`.

Both of these C files get compiled and then archived in
`.pio\build\robotdyn_blackpill_f303cc\libFrameworkHALDriver.a` despite `lib_archive = false` being set in `platformio.ini`. The weak definitions of those 3 functions are then linked into the final binary file.

For a minimal example, I set up a new platformio project for a Robotdyn Blackpill F303CC board (one I have at hand) using the STM32Cube framework. This example program does not function as intended since 3 calls to the STM32Cube HAL framework DAC functions are not linked as expected. My PlatformIO installation is fully up-to-date. Here are the files edited/added:

**platformio.ini**
```ini
[env:robotdyn_blackpill_f303cc]
platform = ststm32
board = robotdyn_blackpill_f303cc
framework = stm32cube
upload_protocol = stlink
debug_tool = stlink
lib_archive = false
```

**main.c** (in `src` folder)
```c
#include "stm32f3xx_hal.h"

DAC_HandleTypeDef hdac;

void DAC_Init(void);
void GPIO_Init(void);
void Error_Handler(void);

int main(void)
{
  uint8_t i = 0;

  HAL_Init();
  DAC_Init();
  GPIO_Init();
  
  HAL_DAC_Start(&hdac, DAC_CHANNEL_1); // Weak definition linked
  while (1)
  {
    HAL_GPIO_TogglePin(GPIOC, GPIO_PIN_13);
    HAL_DAC_SetValue(&hdac, DAC_CHANNEL_1, DAC_ALIGN_8B_R, i); // Weak definition linked
    i += 5;
    HAL_Delay(500);
  }
}

void DAC_Init(void)
{
  DAC_ChannelConfTypeDef sConfig = {0};
  hdac.Instance = DAC;
  if (HAL_DAC_Init(&hdac) != HAL_OK)
  {
    Error_Handler();
  }
  /** DAC channel OUT1 config */
  sConfig.DAC_Trigger = DAC_TRIGGER_NONE;
  sConfig.DAC_OutputBuffer = DAC_OUTPUTBUFFER_ENABLE;
  if (HAL_DAC_ConfigChannel(&hdac, &sConfig, DAC_CHANNEL_1) != HAL_OK) // Weak definition linked
  {
    Error_Handler();
  }
}

void GPIO_Init(void)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  __HAL_RCC_GPIOC_CLK_ENABLE();
  GPIO_InitStruct.Pin = GPIO_PIN_13;
  GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_InitStruct.Pull = GPIO_NOPULL;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_LOW;
  HAL_GPIO_Init(GPIOC, &GPIO_InitStruct);
}

void Error_Handler(void)
{
  __disable_irq();
  while (1);
}

void HAL_MspInit(void)
{
  __HAL_RCC_SYSCFG_CLK_ENABLE();
  __HAL_RCC_PWR_CLK_ENABLE();
}

void HAL_DAC_MspInit(DAC_HandleTypeDef* hdac)
{
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  if(hdac->Instance==DAC)
  {
    /* Peripheral clock enable */
    __HAL_RCC_DAC1_CLK_ENABLE();

    __HAL_RCC_GPIOA_CLK_ENABLE();
    /** DAC GPIO Configuration
    PA4     ------> DAC_OUT1
    */
    GPIO_InitStruct.Pin = GPIO_PIN_4;
    GPIO_InitStruct.Mode = GPIO_MODE_ANALOG;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
  }
}

// Systick Interrupt Handler
void SysTick_Handler(void)
{
  HAL_IncTick();
}
```

stm32f3xx_hal_conf.h (in `src` folder)
```c
#ifndef __STM32F3xx_HAL_CONF_H
#define __STM32F3xx_HAL_CONF_H

#define HAL_MODULE_ENABLED
#define HAL_DAC_MODULE_ENABLED
#define HAL_GPIO_MODULE_ENABLED
#define HAL_EXTI_MODULE_ENABLED
#define HAL_DMA_MODULE_ENABLED
#define HAL_RCC_MODULE_ENABLED
#define HAL_FLASH_MODULE_ENABLED
#define HAL_PWR_MODULE_ENABLED
#define HAL_CORTEX_MODULE_ENABLED

#include "stm32f3xx_hal_rcc.h"
#include "stm32f3xx_hal_gpio.h"
#include "stm32f3xx_hal_exti.h"
#include "stm32f3xx_hal_dma.h"
#include "stm32f3xx_hal_cortex.h"
#include "stm32f3xx_hal_dac.h"
#include "stm32f3xx_hal_flash.h"
#include "stm32f3xx_hal_pwr.h"

#define assert_param(expr) ((void)0U)

#endif /* __STM32F3xx_HAL_CONF_H */
```

fpistm, 2021-01-06:
Hi all, 
This is a known issue.
When weak and non-weak function are in the same library, the linker will used the first one it found.
To avoid this you can use --whole-archive or avoid the strong function to be in the same library.
Anyway in the case of the STM32 F3 DAC HAL function it is not easy to solved. I've met this issue on STM32duino as all core files are built as an archive (core.a). As a first workaround I've used whole archive option (https://github.com/stm32duino/Arduino_Core_STM32/commit/2246ef1b8797725f9dee081d887cdf7b41c870cc) but the main disadvantage is the binary is bigger. So as a second step, I've managed to remove all HAL/LL from the core and build them as an Arduino library which is not built as an archive (https://github.com/stm32duino/Arduino_Core_STM32/pull/677)

valeros, 2021-01-06:
Thanks for clarification. I guess the best we can do is to link HAL sources as objects. In some cases that will lead to a bigger binary, but at least it will work as expected. WDYT?

a2k-hanlon, 2021-01-06:
Thank you for the info @fpistm. @valeros, that sounds reasonable to me. Working is better than not working, even if it's not as small a binary!

In that case, how do you propose that be accomplished? From #205 I thought `lib_archive=false` in the project configuration was the answer, but that doesn't seem to be having any effect on the compilation of the framework. Is it is supposed to?

Alternatively, would another solution just be to reorder the object files in the archive of the HAL framework so the non-weak definitions are found by the linker first? That would avoid the problem of increasing binary size because the framework could still be archived.

caiosignor, 2021-01-07:
I think that I've encontered the same problem using the adc with  stm32f373, the quick solution I found was rename 
 `stm32f3xx_hal_adc_ex.c` to  `stm32f3xx_hal_adc.c`

closed 2021-01-08 by valeros

valeros, 2021-01-08:
I pushed a small fix, @a2k-hanlon could you please test your project using the upstream platform from the dev branch? Just update  the "platform" field in your platformio.ini:
```ini
platform = https://github.com/platformio/platform-ststm32.git
```

Thanks!

a2k-hanlon, 2021-01-09:
@valeros Ok, I tried it. The problem is solved. The project built without errors, linking the correct function definitions, and the program functions as expected. Thank you very much for your help and your efforts!

commit a1e85ff3 (2021-01-08): Link STM32Cube HAL files as objects // Resolve #468

This fixes an issue when weak symbols in a static library get linked
in the final firmware even when proper symbols are present.