platformio/platform-ststm32 #470: PlatformIO support for STM32MP1 dev boards

Pulsetronics, 2021-01-11:
I have been using PlatformIO for most of my projects for years, I want to request support for STM32 processor (STM32MP157A-DK1)

sumit404-work, 2021-07-05:
> I have been using PlatformIO for most of my projects for years, I want to request support for STM32 processor (STM32MP157A-DK1)

I would like to vote this up.

I have started working on running samples from zephyr for stm32mp157 but facing issues while starting a debug run for the M4-cortex core. There is some (_not sure what_) dependency on A7 and OpenOCD. I am working with platformIO on vscode in a windows environment.

SumitNandi-eaton, 2021-12-14:
Hi @valeros , can a label of 'board request' be added for this ?

valeros, 2021-12-14:
@SumitNandi-eaton There is no support for STM32MP1 series at all, so it's a feature.