platformio/platform-ststm32 #473: Mbed: can't upload to nucleo boards

sstaub, 2021-01-14:
Using Mbed, when I try to upload I get following output:
[image]

I get a failed but the 1. Upload works and the program runs
After them no more upload is possible
It happens with all nucleo boards
platformio.ini:
```
[env:nucleo_f401re]
platform = ststm32
board = nucleo_f401re
framework = mbed
upload_protocol = stlink
```

valeros, 2021-01-21:
Should be fixed in the latest mbed package. Please run `pio platform update ststm32` in the IDE terminal.

closed 2021-01-21 by valeros

sstaub, 2021-01-21:
No I can upload with the Mbed framework, but with Arduino framework there is no upload via stlink:
`Processing nucleo_f401re (platform: ststm32; board: nucleo_f401re; framework: arduino)
```
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f401re.html
PLATFORM: ST STM32 (11.0.0) > ST Nucleo F401RE
HARDWARE: STM32F401RET6 84MHz, 96KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Ticker> 3.3.0
Building in release mode
Checking size .pio/build/nucleo_f401re/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.9% (used 908 bytes from 98304 bytes)
Flash: [          ]   2.4% (used 12772 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_f401re/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-12:31)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000a00 msp: 0x20018000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
======================================================================== [SUCCESS] Took 2.92 seconds ========================================================================

Terminal will be reused by tasks, press any key to close it.
```