platformio/platform-ststm32 #478: Upload and Debug doesn't work correctly

MasterLufier, 2021-01-20:
### Configuration

**Operating system**: 
Ubuntu 20.04.1 LTS 

**PlatformIO Version** (`platformio --version`): 
PlatformIO Core, version 5.0.4

**Board**:
Nucleo-F429ZI

### Description of problem
I have a project based on PlatformIO and on PlatformIO version 4 on my another laptop with same OS everything work fine.  After install PlatformIO 5.0.4 on new laptop, OpenOCD can't upload firmware to target.

```
$ platformio run -t nobuild -t upload
~/Qt/Projects/corebootloader_mbed$ platformio run -t nobuild -t upload
Processing nucleo_f429zi (platform: ststm32; board: nucleo_f429zi; framework: mbed)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_f429zi/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
```

If I add `connect_assert_srst` to '.platformio/packages/tool-openocd/scripts/target/stm32f4x.cfg' to end of line 68, uploading work fine

```
$ platformio run -t nobuild -t upload
Processing nucleo_f429zi (platform: ststm32; board: nucleo_f429zi; framework: mbed)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_f429zi/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_assert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08009c54 msp: 0x20030000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```
but debugging still doesn't work.
 
#### Steps to Reproduce
The First problem. Inbuilt GDB doesn't find Python libs (if I right understand)
`$.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb-py .pio/build/nucleo_f429zi/firmware.elf`
```
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
ImportError: No module named site
```
Then I try to take gdb from my old laptop. Python work with it, but debugging still not:

`.platformio/packages/toolchain-gccarmnoneeabi-9/bin/arm-none-eabi-gdb-py --version
GNU gdb (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 8.3.0.20190709-git`

**OpenOCD**
```
$.platformio/packages/tool-openocd/bin/openocd -s .platformio/packages/tool-openocd/scripts -c 'gdb_port 3333' -f .platformio/packages/tool-openocd/scripts/interface/stlink.cfg -f .platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J37M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.262426
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
```
**GDB**
```
.platformio/packages/toolchain-gccarmnoneeabi-9/bin/arm-none-eabi-gdb-py .pio/build/nucleo_f429zi/firmware.elf
...
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
0x00000000 in ?? ()
```
**OpenOCD**
```
Info : accepting 'gdb' connection on tcp/3333
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted
Error executing event gdb-attach on target stm32f4x.cpu:

Info : device id = 0x20036419
Warn : STM32 flash size failed, probe inaccurate - assuming 2048k flash
Info : flash size = 2048 kbytes
Info : Dual Bank 2048 kiB STM32F42x/43x/469/479 found
Info : flash size = 512 bytes
Warn : negative reply, retrying
Warn : negative reply, retrying
```
**GDB**
`(gdb) monitor reset halt`

**OpenOCD**
```
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08009c54 msp: 0x20030000
```
**GDB**
```
(gdb) load
Loading section .text, size 0x1473c lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x801473c
Loading section .data, size 0xa10 lma 0x8014744
Start address 0x8009c54, load size 86356
Transfer rate: 22 KB/sec, 8635 bytes/write.
```
**OpenOCD**
```
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08009c54 msp: 0x20030000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08009c54 msp: 0x20030000
```
**GDB**
```
(gdb) b main
Breakpoint 1 at 0x8010b64: file src/main.cpp, line 7.
(gdb) c
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
jtag status contains invalid mode value - communication failure

Program stopped.
Reset_Handler ()
    at .platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/TOOLCHAIN_GCC_ARM/startup_stm32f429xx.S:78
78	  ldr   sp, =_estack       /* set stack pointer */
```
**OpenOCD**
```
Error: jtag status contains invalid mode value - communication failure
Warn : target stm32f4x.cpu is not halted (gdb fileio)
Polling target stm32f4x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect
Error: jtag status contains invalid mode value - communication failure
Polling target stm32f4x.cpu failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
.....
```
### Additional info
I accidentally updated PlatformIO on my old laptop and now debugging does not work on it too.

valeros, 2021-01-20:
Hi @MasterLufier ! The first part with `connect_assert_srst` is more or less undertandable. What is more interesting is that why uploading works with that new reset method and debugging doesn't. Is there any chance that you're reusing JTAG pins somewhere in your application?

Just to clarify, could you please do the same experiments, but using `arm-none-eabi-gdb` (without python) from the `tool-openocd` package? Also, it would be great if you could try the latest OpenOCD v0.10.0-15 package from this repo https://github.com/xpack-dev-tools/openocd-xpack/releases/tag/v0.10.0-15

Thanks!

MasterLufier, 2021-01-20:
@valeros, thank you for your response.
Today I tried with different combination:
- OpenOCD v0.10.0-15 package from your link
- arm-none-eabi-gdb (without python) from the toolchain-gccarmnoneeabi "version": "1.80301.0" (I think you mean it, not tool-openocd)
- Both arm-none-eabi-gdb from toolchain-gccarmnoneeabi  "version": "1.90000.0" which worked on my old laptop early
and all of this do nothing.
And when I build with flag `platform = ststm32@5.6.0` (with framework-mbed 5.51304.190826 (5.13.4)) debugging start without problem.

MasterLufier, 2021-01-21:
Hi @Copper-Bot. Great thank you for your response!
Replace two files is solved my problem!
overriding ` "target.gpio_reset_at_init"`  does not work for me too.

closed 2021-01-21 by MasterLufier