platformio/platform-ststm32 #479: HAL_ADC_Init failed, STM32Cube framework to STM32F103C8

HamzaHajeir, 2021-01-21:
Hi There 😉

I have an issue programming STM32F103 in PlatformIO (in STM32CUBE framework) , Where everything goes seamlessly, until I hit an error while initializing ADC.

MCU : STM32F103C8

Programming using HAL.

The code is pretty much the same as in STM32CubeIDE, which works seamlessly there.

behavior : 
When function HAL_ADC_Init(&hadc1) is called, it fails with error 1.

> Error_Handler called
> ADC_Init failed 1       Error: 1

A demo project is [here](https://github.com/HamzaHajeir/stm32_pio_issue)

For shorted access, This is my init code :

``` cpp
int main()
{
    HAL_Init();
 
    /* Configure the system clock */
    SystemClock_Config();
    MX_USART3_UART_Init();
    MX_GPIO_Init();
    MX_ADC1_Init();
}
 
void SystemClock_Config(void) {
    RCC_OscInitTypeDef RCC_OscInitStruct = {0};
    RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
    RCC_PeriphCLKInitTypeDef PeriphClkInit = {0};
 
    /** Initializes the CPU, AHB and APB busses clocks 
  */
    RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSI|RCC_OSCILLATORTYPE_HSE;
    RCC_OscInitStruct.HSEState = RCC_HSE_ON;
    RCC_OscInitStruct.HSEPredivValue = RCC_HSE_PREDIV_DIV1;
    RCC_OscInitStruct.HSIState = RCC_HSI_ON;
    RCC_OscInitStruct.LSIState = RCC_LSI_ON;
    RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
    RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
    RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL9;
    if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
    {
        Error_Handler();
    }
    /** Initializes the CPU, AHB and APB busses clocks 
  */
    RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
    RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
    RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
    RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
    RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
 
    if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK) {
        Error_Handler();
    }
 
    PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_ADC;
    PeriphClkInit.AdcClockSelection = RCC_ADCPCLK2_DIV6;
    if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK) {
        Error_Handler();
    }
}
 
 
void MX_USART3_UART_Init(void)
{
 
  huart3.Instance = USART3;
  huart3.Init.BaudRate = 115200;
  huart3.Init.WordLength = UART_WORDLENGTH_8B;
  huart3.Init.StopBits = UART_STOPBITS_1;
  huart3.Init.Parity = UART_PARITY_NONE;
  huart3.Init.Mode = UART_MODE_TX_RX;
  huart3.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart3.Init.OverSampling = UART_OVERSAMPLING_16;
  if (HAL_UART_Init(&huart3) != HAL_OK)
  {
    Error_Handler();
  }
 
}
 
void MX_GPIO_Init(void)
{
 
  GPIO_InitTypeDef GPIO_InitStruct = {0};
 
  /* GPIO Ports Clock Enable */
  __HAL_RCC_GPIOC_CLK_ENABLE();
  __HAL_RCC_GPIOD_CLK_ENABLE();
  __HAL_RCC_GPIOA_CLK_ENABLE();
  __HAL_RCC_GPIOB_CLK_ENABLE();
 
}
 
 
void MX_ADC1_Init(void)
{
 
 
  ADC_ChannelConfTypeDef sConfig = {0};
 
 
  /** Common config 
  */
  hadc1.Instance = ADC1;
  hadc1.Init.ScanConvMode = ADC_SCAN_DISABLE;
  hadc1.Init.ContinuousConvMode = DISABLE;
  hadc1.Init.DiscontinuousConvMode = DISABLE;
  hadc1.Init.ExternalTrigConv = ADC_SOFTWARE_START;
  hadc1.Init.DataAlign = ADC_DATAALIGN_RIGHT;
  hadc1.Init.NbrOfConversion = 1;
 
  int ret= HAL_ADC_Init(&hadc1);
  if ( ret != HAL_OK)
  {
    Error_Handler();
    Serial.printf("ADC_Init failed %u\t",ret);
    ret = HAL_ADC_GetError(&hadc1);
    Serial.printf("Error: %u\n",ret);
  }
  /** Configure Regular Channel 
  */
  sConfig.Channel = ADC_CHANNEL_1;
  sConfig.Rank = ADC_REGULAR_RANK_1;
  sConfig.SamplingTime = ADC_SAMPLETIME_1CYCLE_5;
  if (HAL_ADC_ConfigChannel(&hadc1, &sConfig) != HAL_OK)
  {
    Error_Handler();
    Serial.printf("ADC_ConfigChannel failed\n");
  }
 
 
}

```

### What's my Investigation :
1 (in ADC_GetError call) is defined as HAL_ADC_ERROR_INTERNAL
Description :
/*!< ADC IP internal error: if problem of clocking,
enable/disable, erroneous state */
The error comes from this line in stm32f1xx_hal_adc.c inside HAL_ADC_Init() :
https://pastebin.com/EMcGdwre

- I tried building the project with no optimization, `-O0`, code size bloated, but the problem isn't solved yet.

I don't know what's really happening, Any help or guidance ?

Thanks in Advance 🌺

valeros, 2021-01-21:
Hi @HamzaHajeir ! Could you please try to compile your project using the upstream version of the `ststm32` platform, e.g.
```ini
[env:eval_l073z]
platform = https://github.com/platformio/platform-ststm32.git
framework = stm32cube
board = eval_l073z
```

HamzaHajeir, 2021-01-22:
Hi @valeros , I've just tried it, No thing goes different till now. 

my platformio.ini 
``` ini
[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8
framework = stm32cube
build_unflags = -Os
build_flags = -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1
    -O0
    
lib_deps = https://github.com/philbowles/H4

; monitor_port = COM[6]
monitor_speed = 115200
upload_protocol = stlink
; set CKS mcu ID for knock off of STM chip
upload_flags = -c set CPUTAPID 0x2ba01477
lib_ldf_mode = deep
```

valeros, 2021-01-22:
I'm afraid I won't be able to help you here. As you mentioned, `HAL_ADC_ERROR_INTERNAL` may be caused by improper clock configuration. The best you can do is attach a debug probe and look into the ADC1 registers.

HamzaHajeir, 2021-01-22:
That's what confuses me, It's the same of (everything), starting from the clock and other configs. making the project identical.

What I noticed and could be different, is a behavior when trying to connect ST-LinkV2 (clone) to the MCU. 

It goes unsuccessful _except if clicked reset button_ if flashed the STM32CubeIDE binary.
And it connects successfully without clicking button if flashed the PIO binary.

Might uploading the STM32CubeIDE code help to investigate further ? 
What could be different between PIO and STM32CubeIDE build system ?

HamzaHajeir, 2021-01-24:
For a feedback, I found that `HAL_ADC_MspInit` function is not declared in user code. 

The difference is that STM32CubeIDE declares them internally.

closed 2021-01-24 by HamzaHajeir