platformio/platform-ststm32 #480: Can't upload with st-link using Arduino framework

sstaub, 2021-01-21:
see #473

valeros, 2021-01-21:
HI @sstaub ! It seems that due to #473 you need to completely erase the flash memory on your MCU.

sstaub, 2021-01-21:
How to do this? I can't upload anything, without or with st-link.

valeros, 2021-01-21:
You can try something the following CLI command:
```bash
~/.platformio/packages/tool-openocd/bin/openocd -s ~/.platformio/packages/tool-openocd/scripts -f board/st_nucleo_f4.cfg -c "init" -c "halt" -c "wait_halt" -c "stm32f1x mass_erase 0" -c "sleep 200" -c "reset run" -c "shutdown"
```

sstaub, 2021-01-21:
get this message:
```
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 2000 kHz
Info : STLINK V2J27M15 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.271869
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x08001e78 msp: 0x20017f98
invalid command name "stm32f1x"
sstaub@sstaub1 Ticker % 
```

valeros, 2021-01-21:
Sorry my bad, it should be `stm32f4x`
```bash
~/.platformio/packages/tool-openocd/bin/openocd -s ~/.platformio/packages/tool-openocd/scripts -f board/st_nucleo_f4.cfg -c "init" -c "halt" -c "wait_halt" -c "stm32f4xmass_erase 0" -c "sleep 200" -c "reset run" -c "shutdown"
```

sstaub, 2021-01-22:
doesn't work:
```
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 2000 kHz
Info : STLINK V2J27M15 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.269862
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x08001e78 msp: 0x20017f98
invalid command name "stm32f4xmass_erase"
sstaub@sstaub1 Ticker % 
```

valeros, 2021-01-22:
Try this one:
```
~/.platformio/packages/tool-openocd/bin/openocd -s ~/.platformio/packages/tool-openocd/scripts -f board/st_nucleo_f4.cfg -c "init" -c "halt" -c "wait_halt" -c "stm32f4x mass_erase 0" -c "sleep 200" -c "reset run" -c "shutdown"
```

sstaub, 2021-01-22:
Going back to Mbed Studio, build a small program in release mode. Go to PIO and STM32duino:

```
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_f401re/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-12:31)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

```

valeros, 2021-01-22:
Could you open `~/.platformio/packages/tool-openocd/scripts/target/stm32f4x.cfg` and change the line 68 from:
```
reset_config srst_nogate
```
to
```
reset_config srst_nogate connect_assert_srst
```
This way OpenOCD will try to to connect under reset. Also, as a workaround you can specify `upload_protocol = mbed`

sstaub, 2021-01-22:
It is very deep problem and I don't know how causes it, maybe Mbed Studio, maybe PIO. It is very interesting because I can't upload also with the Arduino IDE, the board isn't recognized but found ?! I work on Mac and i tried on WIN10 to connect board with the St-Link Firmware utility, no luck. I can use Mbed without a problem but not STM32duino.

valeros, 2021-01-22:
Also, what is the color of the ST-Link status diode on the board?

sstaub, 2021-01-22:
The workaround `upload_protocol = mbed` uploads but the program  does not start. The ST-Link LED is green, I think it should red.

sstaub, 2021-01-22:
changed the line in the config:

```
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_f401re/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-12:31)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_assert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000a00 msp: 0x20018000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

valeros, 2021-01-22:
OK, seems that it works now? In a nutshell, there is a bug in the latest mbed 6.6.0, so each time you upload firmware from Mbed Studio you brick the board. That's why you need to erase the entire flash memory.

sstaub, 2021-01-22:
No, I can upload, but nothing happens, tried with a simple blinky, no blink. 

```
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_f401re/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-12:31)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_assert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000a00 msp: 0x20018000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

valeros, 2021-01-22:
Does it work in the Arduino IDE?

sstaub, 2021-01-22:
No, it doesn't work, the board is mounted and Arduino recognize the board.
BigSur problem?

```
NODE_F401RE,NOD_F401RE not found.
An error occurred while uploading the sketch
Please ensure the device is correctly connected and mounted.
```

mr3188, 2021-02-03:
I'm having the same issue. 
This worked yesterday without any problem.

After adding connect_assert_srst I get the following output and the board just send a NULL char through serial

```Checking size .pio\build\nucleo_l476rg\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   1.6% (used 2152 bytes from 131072 bytes)
Flash: [          ]   2.8% (used 29320 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\nucleo_l476rg\firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_assert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000b10 msp: 0x20018000
** Programming Started **
Warn : Adding extra erase range, 0x08007438 .. 0x080077ff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
in procedure 'program'
*** [upload] Error 1```

mr3188, 2021-02-03:
I changed platforfmio version to an older one, changed back to the latest, erased the board and now it seems to work again. No idea what really happened.