platformio/platform-ststm32 #481: Expand stm32cube builder script to build CMSIS-RTOS2 component

maxgerhardt, 2021-01-22:
Per https://community.platformio.org/t/cant-find-cmsis-rtos2-on-nucleo-l452re-with-stm32cube-framework/18713.

The current builder script only uses a few things from the Drivers/CMSIS folder of the STM32Cube package, namely "Device" and "DSP". 

https://github.com/platformio/platform-ststm32/blob/a1e85ff37ced73048b23b3c99eba6b1b0c6f9dc4/builder/frameworks/stm32cube.py#L202-L227

It would be nice to also configurably be able to build the other components in there, namely CMSIS-RTOS2 (I'm ignoring the older CMSIS-RTOS here) and maybe CMSIS-NN for neural networks.

maxgerhardt, 2021-01-22:
Problem: The STM32Cube package doesn't contain the actual core code (RTX) from CMSIS-RTOS2! 

[image]

Compare to https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/RTOS2, the entire RTX folder is missing. 

Is this due to licensing issues?

tcaputi, 2021-01-22:
The code that STMCubeMx generates also doesn't include that. I think that it tries to use FreeRTOS for all of the internals. I believe all the code is there, its just not setup to build that code. I could be wrong though.

valeros, 2021-02-04:
For some reason they don't include that files in the official STM32Cube packages (e.g. https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Drivers/CMSIS/RTOS2) even though the RTX RTOS has a permissive license.

caiosignor, 2021-02-04:
Oh, I was trying it yesterday, but with rtos1