platformio/platform-ststm32 #485: ModuleNotFoundError: No module named 'tools.build_api'

anhduong1512, 2021-01-26:
PIO Core Call Error: "The current working directory C:\\Users\\Anh Duong\\Documents\\PlatformIO\\Projects\\P1 will be used for the project.\r\n\r\nThe next files/directories have been created in C:\\Users\\Anh Duong\\Documents\\PlatformIO\\Projects\\P1\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Processing nucleo_l152re (platform: ststm32; board: nucleo_l152re; framework: mbed)\r\n--------------------------------------------------------------------------------\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_l152re.html\r\nPLATFORM: ST STM32 (10.0.0) > ST Nucleo L152RE\r\nHARDWARE: STM32L152RET6 32MHz, 80KB RAM, 512KB Flash\r\nDEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)\r\nPACKAGES: \r\n - framework-mbed 6.60200.200722 (6.2.0) \r\n - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)\r\nModuleNotFoundError: No module named 'tools.build_api':\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 170:\r\n    env.SConscript(\"$BUILD_SCRIPT\")\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 598:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 287:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\platforms\\ststm32\\builder\\main.py\", line 99:\r\n    target_elf = env.BuildProgram()\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 61:\r\n    env.ProcessProgramDeps()\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 127:\r\n    env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Environment.py\", line 219:\r\n    return self.method(*nargs, **kwargs)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 342:\r\n    SConscript(env.GetFrameworkScript(f), exports=\"env\")\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 661:\r\n    return method(*args, **kw)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 598:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 287:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\platforms\\ststm32\\builder\\frameworks\\mbed.py\", line 37:\r\n    \"platformio-build.py\"))\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 661:\r\n    return method(*args, **kw)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 598:\r\n    return _SConscript(self.fs, *files, **subst_kw)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\tool-scons\\scons-local-4.0.1\\SCons\\Script\\SConscript.py\", line 287:\r\n    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\framework-mbed\\platformio\\platformio-build.py\", line 43:\r\n    from pio_mbed_adapter import PlatformioMbedAdapter\r\n  File \"C:\\Users\\Anh Duong\\.platformio\\packages\\framework-mbed\\platformio\\pio_mbed_adapter.py\", line 23:\r\n    from tools.build_api import prepare_toolchain, UPDATE_WHITELIST\r\n========================== [FAILED] Took 0.84 seconds =========================="

valeros, 2021-01-26:
Hi @anhduong1512 ! Could you try to reproduce the using using the latest version of the `ststm32` platform?

dwhall, 2024-07-05:
I have a project that gives me the `ModuleNotFoundError: No module named 'tools.build_api'`.  I hope you can use it to recreate and resolve the issue.  

> git clone https://github.com/dwhall/nim-pio-template.git 

To reproduce the exception, build env `nucleo_f446re-mbed`.  
Note that env `nucleo_f446re` (framework=cmsis) builds without the exception.
My system is
Windows 10 Home build 19045.4529
PlatformIO: Core 6.1.16a1·Home 3.4.4
ST STM32 17.4.0

I've attached my build output [build_output.txt](https://github.com/user-attachments/files/16104878/build_output.txt)