platformio/platform-ststm32 #490: Can not get a successful build with STM32CubeMX and USB lib

Logiase, 2021-01-30:
## Reproduction

- `genericSTM32F103C8` board
- `stm32cube` framework
- using `stm32pio` to generate project
- choose any of `USB_FS` in `middleware` in `STM32CubeMX`

and build.

## Reason

template file in the `STM32_USB_Device_Library/Class` can not lead to a successful build.

## Solution

after knowing the reason, it's a quite easy to solve.

The temporary solution is rename the template file, for example, `usbd_customhid_if_template.c` to `usbd_customhid_if_template.c.template`.

Maybe, this bug exists on all `stm32cube` framework.

valeros, 2021-02-03:
Hi @Logiase ! Could you please compile your project in verbose mode and attach here the log?

Logiase, 2021-02-03:
> Hi @Logiase ! Could you please compile your project in verbose mode and attach here the log?

Log:

```text
Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: stm32cube)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 (11.0.0) > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-stm32cubef1 1.8.3
 - tool-ldscripts-ststm32 0.1.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find a linker script for the required board! An auto-generated script will be used to link firmware!
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...     
Dependency Graph
|-- <STM32_USB_Device_Library-Core> (C:\Users\Logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core)
|-- <STM32_USB_Device_Library-CustomHID> (C:\Users\Logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID)
|   |-- <STM32_USB_Device_Library-Core> (C:\Users\Logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core)
Building in release mode
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\src\gpio.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DPLATFORMIO=50100 -DSTM32F103xB -DSTM32F1 -DUSE_HAL_DRIVER -DF_CPU=72000000L -IInc -ISrc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Src -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Src -ISrc -IInc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\CMSIS\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\CMSIS\Device\ST\STM32F1xx\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Src Src\gpio.c
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\src\main.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DPLATFORMIO=50100 -DSTM32F103xB -DSTM32F1 -DUSE_HAL_DRIVER -DF_CPU=72000000L -IInc -ISrc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Src -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Src -ISrc -IInc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\CMSIS\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\CMSIS\Device\ST\STM32F1xx\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Src Src\main.c
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\src\stm32f1xx_hal_msp.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DPLATFORMIO=50100 -DSTM32F103xB -DSTM32F1 -DUSE_HAL_DRIVER -DF_CPU=72000000L -IInc -ISrc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Src -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Src -ISrc -IInc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\CMSIS\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\CMSIS\Device\ST\STM32F1xx\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Src Src\stm32f1xx_hal_msp.c
arm-none-eabi-gcc -o .pio\build\genericSTM32F103C8\src\stm32f1xx_it.o -c -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m3 -nostdlib -DPLATFORMIO=50100 -DSTM32F103xB -DSTM32F1 -DUSE_HAL_DRIVER -DF_CPU=72000000L -IInc -ISrc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Class\CustomHID\Src -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Middlewares\ST\STM32_USB_Device_Library\Core\Inc -IC:\users\logias [... cut ...] \Device\ST\STM32F1xx\Include -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Inc -IC:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Src C:\users\logiase\.platformio\packages\framework-stm32cubef1\Drivers\STM32F1xx_HAL_Driver\Src\stm32f1xx_hal_can.c
==================================================================== [FAILED] Took 6.04 seconds ====================================================================
```

valeros, 2021-02-03:
Many thanks for reporting, should be fixed in the dev branch, please retest with the upstream version of the platform:
```ini
[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8
framework = stm32cube
```

commit 2f2cc611 (2021-02-03): Fix source filter template for internal stm32cube libs
Resolve #490
closed 2021-02-03 by valeros