platformio/platform-ststm32 #493: problem with stm32 bluePillf103C8 generic

alexgalstyan, 2021-02-02:
In IPAddress class declared ing IPAddress.h
after update at 02.02.20201.
the class is missing a function toString().

error in my code 

error: 'class IPAddress' has no member named 'toString'; did you mean 'fromString'?

is it OK?

maxgerhardt, 2021-02-02:
Whichcore are you using? https://github.com/stm32duino/Arduino_Core_STM32/tree/master/cores/arduino? If they removed code that you need, you need to go there. 

Otherwise try and find an alternative. First, since `IPAddress` implements `Printable` you can directly pass it to the `print(ln)` function of e.g. a `Serial`. Second you can write a quick conversion function with the same code as in `IPAddress:printTo`

```cpp
String ipaddr_to_str(const IPAddress& address)
{
  String output = "";
  for (int i = 0; i < 3; i++)
  {
    output += String(address[i], DEC);
    output += ".";
  }
  output += String(address[3], DEC);
  return output;
}
```

Second, if you need an older core version, you can specify that with [`platform_packages`](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages) and a version for your `framework-arduinoststm32` or `framework-arduinoststm32-maple` package at https://bintray.com/platformio/tool-packages.

fpistm, 2021-02-02:
Hi,
`toString `has never been part of IPAddress in the STM32 core. But it is part of the LibMaple core.
My guess is that now the STM32 core is used by default for all STM32 Arduino target you get this issue.
You could specify in your json to use the maple core instead of the default one.

maxgerhardt, 2021-02-02:
Ah, indeed the maple core has it, not the STM32 core 

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/cores/maple/IPAddress.h#L67

So it should suffice to just add

```
board_build.core = maple
```
to the `platformio.ini`

alexgalstyan, 2021-02-03:
> Which core are you using? https://github.com/stm32duino/Arduino_Core_STM32/tree/master/cores/arduino? If they removed code that you need, you need to go there.
> 
> Otherwise try and find an alternative. First, since `IPAddress` implements `Printable` you can directly pass it to the `print(ln)` function of e.g. a `Serial`. Second you can write a quick conversion function with the same code as in `IPAddress:printTo`
> 
> ```c++
> String ipaddr_to_str(const IPAddress& address)
> {
>   String output = "";
>   for (int i = 0; i < 3; i++)
>   {
>     output += String(address[i], DEC);
>     output += ".";
>   }
>   output += String(address[3], DEC);
>   return output;
> }
> ```
> 
> Second, if you need an older core version, you can specify that with [`platform_packages`](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages) and a version for your `framework-arduinoststm32` or `framework-arduinoststm32-maple` package at https://bintray.com/platformio/tool-packages.

alexgalstyan, 2021-02-03:
> In IPAddress class declared ing IPAddress.h
> after update at 02.02.20201.
> the class is missing a function toString().
> 
> error in my code
> 
> error: 'class IPAddress' has no member named 'toString'; did you mean 'fromString'?
> 
> is it OK?

Thank you

alexgalstyan, 2021-02-03:
> Hi,
> `toString `has never been part of IPAddress in the STM32 core. But it is part of the LibMaple core.
> My guess is that now the STM32 core is used by default for all STM32 Arduino target then you get this issue.
> You could specify in your json to use the maple core instead of the default one.

15 day ago it was :)

alexgalstyan, 2021-02-03:
> Ah, indeed the maple core has it, not the STM32 core
> 
> https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/cores/maple/IPAddress.h#L67
> 
> So it should suffice to just add
> 
> ```
> board_build.core = maple
> ```
> 
> to the `platformio.ini`

it is not help.

alexgalstyan, 2021-02-03:
My platformio.ini content
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
debug_tool = stlink

Of course I'm using my conversion function now. But 2 weeks ago there was no such need.
I just asked for notice.

maxgerhardt, 2021-02-03:
>It did not help

Yes it helps. I personally tested it.

[image]

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
board_build.core = maple
```

```cpp
#include <Arduino.h>
#include <IPAddress.h>

void setup() {
  IPAddress ip(192,168,0,1);
  String asStr = ip.toString();
  Serial.println(asStr);
}

void loop() {}
```

>15 day ago it was :)

@fpistm is right, the **STM32Duino core** (or STM32 core called in PlatformIO), never contained the `toString()` function. This is publically verifyable in the history of the file.

https://github.com/stm32duino/Arduino_Core_STM32/commits/master/cores/arduino/IPAddress.h

On the other han only the **STM32 Maple core** every had this function, as linked above. 

Since release [7.0.0](https://github.com/platformio/platform-ststm32/releases), the Maple boards were switched to use the STM32Duino core by default. So if you updated your platform, that would be the reason why it does not compile anymore.

As I have shown above, you can still switch to the maple core as I've show, and compilation works.

If the code does not compile for you given my `platformio.ini`, please post a compilation of (output of "Build").

alexgalstyan, 2021-02-03:
Sorry, Yes it helps.

closed 2021-02-03 by alexgalstyan