platformio/platform-ststm32 #496: Request rupport for NUCLEO-G491RE with STM32G491RE

Gamester17, 2021-02-16:
Requesting support NUCLEO-G491RE with STM32G491RE

-https://www.st.com/en/evaluation-tools/nucleo-g474re.html
    - https://www.st.com/en/microcontrollers-microprocessors/stm32g474re.html

That is almost same as NUCLEO-G474RE and NUCLEO-G431RB devboards with different MCU that are already [STSTM32](https://github.com/platformio/platform-ststm32) supports:

https://docs.platformio.org/en/latest/boards/ststm32/nucleo_g474re.html

https://docs.platformio.org/en/latest/boards/ststm32/nucleo_g431rb.html

https://github.com/platformio/platform-ststm32/blob/252752d7c2745e387862d11f26f35cf2180f5ead/boards/nucleo_g474re.json

https://github.com/platformio/platform-ststm32/blob/252752d7c2745e387862d11f26f35cf2180f5ead/boards/nucleo_g431rb.json

PS: All of these STM32G4 devboards use STMicro's STM32 Nucleo-64 MB1367 reference board form factor and pin layout.

GunZi200, 2021-04-02:
@valeros there is no `.svd` file for `STM32G491xx`. Are they pulled from somewhere else or do we need to manually add it?

GunZi200, 2021-04-02:
I updated all the SVD files here: https://github.com/GunZi200/platform-ststm32/tree/svd-update I can create a PR for this. Many new files and many SVD files are updated

gudnimg, 2021-06-13:
@Gamester17 I created a PR for this, it would be great if you could test it? :)