platformio/platform-ststm32 #500: Could not load installed platforms

heman22union, 2021-03-03:
Hello,

I just installed the Platform IO vscode extension and the auto build marlin extension so I can update my ender 5 pro 3d printer. When using the auto build marlin extension, I get errors on the console saying unable to locate ststm32 platform. 

I cloned the marlin bugfix 2.0.x repo and their platform.io appears to be good.

Any suggestions on what to do? I saw the same exact error message listed in several closed issues on the platformio-home repo.

heman22union, 2021-03-03:
I removed the stuff from the ststm32 platform and used the auto build marlin extension again. Here is the terminal output.

Heman@XPS-13 MINGW64 ~/Code/Marlin (2.0.x)
$ platformio run -e STM32F103RET6_creality
Processing STM32F103RET6_creality (platform: ststm32@~6.1; board: genericSTM32F103RE; framework: arduino)
arduino)                                                                                         --------------------------------------------------------------------------------------------------------------------------------
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing ststm32 @ ~6.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: ststm32 @ 6.1.1 has been installed!
The platform 'ststm32@~6.1' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.            01.0,<1.80000.0 requirements:
Tool Manager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0
Tool Manager: Warning! More than one package has been found by toolchain-gccarmnoneeabi @ >=1.60301.0,<1.80000.0 requirements:                                                                     to avoid name conflicts     
 - platformio/toolchain-gccarmnoneeabi @ 1.50401.210106
 - maxgerhardt/toolchain-gccarmnoneeabi @ 1.40903.0
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: toolchain-gccarmnoneeabi @ 1.70201.0 has been installed!
Tool Manager: Installing framework-arduinoststm32-maple @ ~2.10000.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoststm32-maple @ 2.10000.200103 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (6.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0)
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing SoftwareSerialM
Library Manager: SoftwareSerialM @ 1.0.0 has been installed!
Library Manager: Installing U8glib-HAL @ ~0.4.1
Library Manager: Warning! More than one package has been found by U8glib-HAL @ ~0.4.1 requirements:
 - marlinfirmware/U8glib-HAL @ 0.4.3
 - thinkyhead/U8glib-HAL @ 0.4.3
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: U8glib-HAL @ 0.4.3 has been installed!
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
*** [C:\Users\Heman\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103r\board.cpp] C:\Users\Heman\Code\Marlin\.pio\build\STM32F103RET6_creality\.sconsign37.dblite: No such file or directory
================================= [FAILED] Took 2313.60 seconds =================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:38:33.603
============================= 1 failed, 0 succeeded in 00:38:33.603 =============================

Heman@XPS-13 MINGW64 ~/Code/Marlin (2.0.x)
$ echo "done" >C:\Users\Heman\AppData\Local\Temp\ipc

Heman@XPS-13 MINGW64 ~/Code/Marlin (2.0.x)

heman22union, 2021-03-03:
Here is a portion of the platform.io file that was in the Marlin repo.

#
# Creality (STM32F103RET6)
#
[env:STM32F103RET6_creality]
platform        = ${common_stm32f1.platform}
extends         = common_stm32f1
board           = genericSTM32F103RE
build_flags     = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags} -std=gnu++14 -DSTM32_XL_DENSITY -DTEMP_TIMER_CHAN=4
extra_scripts   = ${common.extra_scripts}
  pre:buildroot/share/PlatformIO/scripts/random-bin.py
  buildroot/share/PlatformIO/scripts/creality.py
lib_ignore      = ${common_stm32f1.lib_ignore}
debug_tool      = jlink
upload_protocol = jlink
monitor_speed   = 115200

valeros, 2021-03-03:
Is there a whitespace character in your project path?

heman22union, 2021-03-03:
> Is there a whitespace character in your project path?

Heman@XPS-13 MINGW64 ~/Code/Marlin (2.0.x)
$ pwd
/c/Users/Heman/Code/Marlin

I don't think so..?

This is output form git for windows on the BASH terminal

heman22union, 2021-03-03:
I just tried a build again. Might be getting somewhere. I'm going to review and get back.

valeros, 2021-03-03:
I have a suspicion that the Git terminal is messing with paths, I'd recommend you to run the project using the native CMD.