platformio/platform-ststm32 #504: FileNotFoundError with debug

pat1, 2021-03-17:
```
pio system info
--------------------------  ----------------------------------------------------
PlatformIO Core             5.1.0
Python                      3.8.7-final.0
System Type                 linux_x86_64
Platform                    Linux-5.10.20-100.fc32.x86_64-x86_64-with-glibc2.2.5
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/pat1/.platformio
PlatformIO Core Executable  /home/pat1/.platformio/penv/bin/platformio
Python Executable           /home/pat1/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       3
Tools & Toolchains          28
--------------------------  ----------------------------------------------------
```
platformio.ini:
```
[env:nucleo_l432kc]
framework = arduino
platform = ststm32
board = nucleo_l432kc

[env:microduino_stm32f103]
framework = arduino
platform = ststm32
board = genericSTM32F103CB
board_build.variants_dir = ../../custom_variants
board_build.variant = microduinostm32_F103CB
build_flags =
	-DUSBCON
	-DUSBD_VID=0
	-DUSBD_PID=0
	-DHAL_PCD_MODULE_ENABLED
	-DUSBD_USE_CDC
upload_protocol = stlink
debug_tool = stlink
```
```
pio debug --interface=gdb 
Warning! Your `/etc/udev/rules.d/99-platformio-udev.rules` are outdated. Please update or reinstall them.
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules
Preparing firmware for debugging...
Processing nucleo_l432kc (framework: arduino; platform: ststm32; board: nucleo_l432kc)
-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_l432kc.html
PLATFORM: ST STM32 (12.0.0) > ST Nucleo L432KC
HARDWARE: STM32L432KCU6 80MHz, 64KB RAM, 256KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting test_sdcard.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <SdFat> 2.0.5
|   |-- <SPI> 1.0
Building in debug mode
Compiling .pio/build/nucleo_l432kc/src/test_sdcard.ino.cpp.o
Checking size .pio/build/nucleo_l432kc/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.6% (used 2380 bytes from 65536 bytes)
Flash: [=         ]  11.5% (used 30160 bytes from 262144 bytes)
========================================================= [SUCCESS] Took 1.89 seconds =========================================================

Environment    Status    Duration
-------------  --------  ------------
nucleo_l432kc  SUCCESS   00:00:01.889
========================================================= 1 succeeded in 00:00:01.889 =========================================================
Upon execvpe arm-none-eabi-gdb [b'piogdb', b'-q', b'--directory', b'/home/pat1/.platformio/.cache/.piodebug-269q69rg', b'--directory', b'/home/pat1/git/rmap/platformio/stima_v3/test_sdcard', b'-l', b'10', b'/home/pat1/git/rmap/platformio/stima_v3/test_sdcard/.pio/build/nucleo_l432kc/firmware.elf'] in environment id 139674608843904
:Traceback (most recent call last):
  File "/home/pat1/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 405, in _fork
    self._execChild(path, uid, gid, executable, args,
  File "/home/pat1/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 484, in _execChild
    os.execvpe(executable, args, environment)
  File "/usr/lib64/python3.8/os.py", line 577, in execvpe
    _execvpe(file, args, env)
  File "/usr/lib64/python3.8/os.py", line 610, in _execvpe
    raise last_exc
  File "/usr/lib64/python3.8/os.py", line 601, in _execvpe
    exec_func(fullname, *argrest)
FileNotFoundError: [Errno 2] No such file or directory: b'/usr/libexec/sdcc/arm-none-eabi-gdb'

```

with atom I get:

Error launching PIO Debugging Server
```
Upon execvpe arm-none-eabi-gdb [b'piogdb', b'-q', b'--directory', b'/home/pat1/.platformio/.cache/.piodebug-k1l8w8ly', b'--directory', b'/home/pat1/git/rmap/platformio/stima_v3/test_sdcard', b'-l', b'10', b'--interpreter=mi', b'-nx', b'/home/pat1/git/rmap/platformio/stima_v3/test_sdcard/.pio/build/nucleo_l432kc/firmware.elf'] in environment id 140594451226304 :Traceback (most recent call last): File "/home/pat1/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 405, in _fork self._execChild(path, uid, gid, executable, args, File "/home/pat1/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 484, in _execChild os.execvpe(executable, args, environment) File "/usr/lib64/python3.8/os.py", line 577, in execvpe _execvpe(file, args, env) File "/usr/lib64/python3.8/os.py", line 610, in _execvpe raise last_exc File "/usr/lib64/python3.8/os.py", line 601, in _execvpe exec_func(fullname, *argrest) FileNotFoundError: [Errno 2] No such file or directory: b'/home/pat1/.platformio/penv/bin/arm-none-eabi-gdb'
```

ivankravets, 2021-03-17:
> with atom I get:

Atom is deprecated. Please switch to VSCode. https://docs.platformio.org/en/latest/integration/ide/vscode.html#ide-vscode

pat1, 2021-03-18:
ok for deprecated Atom, not so ok for VSCode (license), but it do not work for console too

pio debug --interface=gdb 

as show on my previous post

ivankravets, 2021-03-18:
Could you try the latest PIO Core 5.2 beta? `pio upgrade --dev`. Does it work? We refactored debugging component.

pat1, 2021-03-18:
```
pio upgrade --dev
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 5.2.0a1
Release notes: https://docs.platformio.org/en/latest/history.html
[pat1@localhost test_sdcard]$ pio debug --interface=gdb
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 5.2.0a1!

***********************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
***********************************************************************************************************************************************************************************************************************

Warning! Your `/etc/udev/rules.d/99-platformio-udev.rules` are outdated. Please update or reinstall them.
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules
Preparing firmware for debugging...
Processing microduino_stm32f103 (platform: ststm32; board: genericSTM32F103CB; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103CB.html
PLATFORM: ST STM32 (12.0.0) > STM32F103CB (20k RAM. 128k Flash)
HARDWARE: STM32F103CBT6 72MHz, 20KB RAM, 128KB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting test_sdcard.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI> 1.0
|-- <SdFat> 2.0.5
|   |-- <SPI> 1.0
Building in debug mode
....
Linking .pio/build/microduino_stm32f103/firmware.elf
Checking size .pio/build/microduino_stm32f103/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  25.6% (used 5244 bytes from 20480 bytes)
Flash: [===       ]  30.4% (used 39792 bytes from 131072 bytes)
Building .pio/build/microduino_stm32f103/firmware.bin
============================================================================================= [SUCCESS] Took 6.89 seconds =============================================================================================

Environment           Status    Duration
--------------------  --------  ------------
microduino_stm32f103  SUCCESS   00:00:06.895
============================================================================================= 1 succeeded in 00:00:06.895 =============================================================================================
Error: Traceback (most recent call last):
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/platformio/__main__.py", line 115, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pat1/.platformio/penv/lib64/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
[... 12 lines cut ...]
  File "/usr/lib64/python3.8/asyncio/base_subprocess.py", line 36, in __init__
    self._start(args=args, shell=shell, stdin=stdin, stdout=stdout,
  File "/usr/lib64/python3.8/asyncio/unix_events.py", line 789, in _start
    self._proc = subprocess.Popen(
  File "/usr/lib64/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib64/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'arm-none-eabi-gdb'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

pat1, 2021-03-19:
Same problem with vscode.

 The problem seems due to a missed arm-none-eabi-gdb system installed.

ln -s /home/pat1/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/arm-none-eabi-gdb /usr/bin/arm-none-eabi-gdb

solve the problem  for command line and Atom too.
So I think platformio need to use correct path for gdb command.