platformio/platform-ststm32 #506: Running code from RAM

erdnaxe, 2021-03-18:
I would like to contribute a feature to enable people **to run code from memory rather than flash**.
I believe this feature already exist in STM32CubeIDE when "debugging from RAM".

This feature is useful in some rare cases such as side channel evaluations in which we do not want the flash to be accessed. It can also be useful in a setup in which we flash the target 1000 times per day to preserve flash longevity.

From what I have seen, I believe to implement this, we need to:

- Change ld script to put code in RAM.
- Change STLink/JLink script to move PC to the correct address.

Should this feature be contributed in this repository? Do you have advice on where to begin?

Best regards,

valeros, 2021-03-23:
Hi @erdnaxe ! How does this workflow look like in the STM32CubeIDE? IMO, it's much easier to adjust your project configuration instead of implementing this feature directly in the platform as it doesn't seem to be  worth the effort.

erdnaxe, 2021-03-26:
Hi! In STM32CubeIDE by default two files `STM32F030R8TX_FLASH.ld` and `STM32F030R8TX_RAM.ld` are generated (on STM32F4 at least). Then using the debug configuration GUI it's possible to choose between the two.

Because I'm using LibOpenCM3, I am following <https://github.com/libopencm3/libopencm3/wiki/Run-From-RAM>, so actually this translates in `platformio.ini` by:

```ini
; Disable libopencm3 ldscript generation and provide custom one
board_build.ldscript = ${PROJECT_DIR}/stm32f0.ld"
```

with `stm32f0.ld` (for one STM32F0 board) being:

```c
MEMORY
{
        rom (rx) : ORIGIN = 0x20000000, LENGTH = 4K
        ram (rwx) : ORIGIN = 0x20001000, LENGTH = 4K  // was 8K
}

INCLUDE cortex-m-generic.ld
```

I am going to test this setup, if this is working maybe I could document this somewhere.