platformio/platform-ststm32 #511: PortentaH7 upload issue *** [upload] Error 74

JLH-94, 2021-03-31:
Hello,
i have an PortentaH7-board conntected via USB to a windows 10 computer. 
If i upload the example project to the M7-core in the arduino IDE it works.
In VScode with PlatformIO i gets the following error message. (see below)
Are there already solutions, I know that the PortentaH7 was just added.

> Executing task in folder Test_PortentaH7_M7core: C:\Users\jh\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing portenta_h7_m7 (platform: ststm32; board: portenta_h7_m7; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m7.html
PLATFORM: ST STM32 (12.1.0) > Arduino Portenta H7 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduino-mbed 1.3.2
 - tool-dfuutil 1.9.200310
 - tool-openocd 2.1000.200630 (10.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
Building in release mode
Checking size .pio\build\portenta_h7_m7\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.8% (used 66984 bytes from 523624 bytes)
Flash: [===       ]  34.5% (used 271256 bytes from 786432 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Auto-detected: COM4
Uploading .pio\build\portenta_h7_m7\firmware.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
No DFU capable USB device available
*** [upload] Error 74
================================================================================================= [FAILED] Took 1.27 seconds =================================================================================================The terminal process "C:\Users\jh\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

**Just for reference ArduinoIDE output:**
(Also some DFU problems, already read that its maybe about permissions but the upload works just fine anyways)

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 2341:035b
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08040000, size = 132976
Download	[=========================] 100%       132976 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Cannot open DFU device 2341:035b

maxgerhardt, 2021-03-31:
Can you please edit the issue text to use [Markdown code formatting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with tripple backtick to format it correctly? 

Also the invocation line in the Arduino IDE is 

>C:\Users\Max\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.9.0-arduino2/dfu-util --device 0x2341:0x035b -D C:\Users\Max\AppData\Local\Temp\arduino_build_767468/sketch_mar31a.ino.bin -a0 --dfuse-address=0x08040000:leave 

Whereas in PlatformIO 

>... I can show this after PlatformIO has finished downloading framework-arduino-mbed for 8 minues... 

Maybe also a build step is missing here to create DFU binaries? @valeros

JLH-94, 2021-03-31:
I edited the issue text, thanks for that hint.
In the meantime, I also tried uploading to M4 Core with the same result as expected.

maxgerhardt, 2021-03-31:
Does the behavior change in any way if you add

```
upload_protocol = dfu-util
```
to the `platformio.ini`?

JLH-94, 2021-03-31:
No unfortunately not, i compared the consol output same result.

valeros, 2021-04-01:
It seems that there is something wrong with the DFU header, maybe the Arduino team uses a custom DFU tool. @JLH-94 could please run with the following command (please build the PlatformIO project before running the command) just to make sure that the generated binary is correct and can be uploaded:

```
C:\Users\Max\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.9.0-arduino2/dfu-util --device 0x2341:0x035b -D C:\Users\Max\Documents\PlatformIO\Projects\test_proj_3\.pio\build\portenta_h7_m7\firmware.bin -a0 --dfuse-address=0x08040000:leave
```

maxgerhardt, 2021-04-01:
*With corrected username for his Windows account :D.

datacode-taavi, 2021-04-01:
I have the same problem. Uploading works correctly with Arduino IDE, however PIO fails to upload.

If I set the device to bootloader mode and then attempt to upload using Arduino DFU tool, then the upload succeeds (in default mode it fails, even if I change the USB Product ID):
```
/home/taavi/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino2/dfu-util --device 0x2341:0x035b -D .pio/build/portenta_h7m7/firmware.bin -a0 --dfuse-address=0x08040000:leave
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2341:035b
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08040000, size = 114592
Download	[=========================] 100%       114592 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

With Platform IO, I fail to get it uploaded no matter what state the device is in:
```
$ pio run -e portenta_h7m7 -t upload -v
Processing portenta_h7m7 (platform: ststm32; board: portenta_h7_m7; framework: arduino; upload_protocol: dfu)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m7.html
PLATFORM: ST STM32 (12.1.0) > Arduino Portenta H7 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed 1.3.2 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library /home/taavi/.platformio/packages/framework-arduino-mbed/libraries/LittleVGL
Framework incompatible library /home/taavi/.platformio/packages/framework-arduino-mbed/libraries/Portenta_SDCARD
Framework incompatible library /home/taavi/.platformio/packages/framework-arduino-mbed/libraries/mbed-memory-status
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 24 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/portenta_h7m7/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.5% (used 55176 bytes from 523624 bytes)
Flash: [=         ]  14.5% (used 114072 bytes from 786432 bytes)
.pio/build/portenta_h7m7/firmware.elf  :
section               size        addr
.text               111640   134479872
.ARM.extab             520   134591512
.ARM.exidx            1032   134592032
.crash_data_ram        256   603980440
.data                 1400   603980696
.bss                 53776   603982096
.heap               467168   604035872
.openamp_section         0   939524096
.lwip_sec           278528   805306368
.ARM.attributes         44           0
.comment               126           0
.debug_info        2722120           0
.debug_abbrev       237710           0
.debug_aranges       26496           0
.debug_ranges        42584           0
.debug_line         372338           0
.debug_str          750637           0
.debug_frame         83932           0
.debug_loc          331401           0
.stab                   60           0
.stabstr               118           0
Total              5481886
<lambda>(["upload"], [".pio/build/portenta_h7m7/firmware.bin"])
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pio/build/portenta_h7m7/firmware.bin"])
Auto-detected: /dev/ttyACM0
"/home/taavi/.platformio/packages/tool-dfuutil/bin/dfu-util" -d 0x2341:0x035b,0x2341:0x025b,0x2341:0x045b -a 0 -s 0x08040000:leave -D "/home/taavi/CLionProjects/portenta_test/.pio/build/portenta_h7m7/firmware.bin"
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
*** [upload] Error 74
========================================================================================================= [FAILED] Took 0.44 seconds =========================================================================================================
```

valeros, 2021-04-01:
Can someone try the latest platform revision from the dev branch? 
```ini
[env:portenta_h7_m4]
platform = https://github.com/platformio/platform-ststm32.git
framework = arduino
board = portenta_h7_m4
```

maxgerhardt, 2021-04-01:
>If I set the device to bootloader mode and then attempt to upload using Arduino DFU tool, then the upload succeeds (in default mode it fails, even if I change the USB Product ID):

That part is expected because the device is reset into bootloader mode by a special reset procedure (opening the COM port at 1200bps), and if you execute the command directly that's not done before. 

Maybe it works with the updated platform code now? (I don't have the board)

JLH-94, 2021-04-01:
So I tested the suggestion with the command input. And indeed it works perfectly with it. After setting the PortentaH7 in boot mode you can upload the project with this command.

```
PS C:\Users\jh\Documents\PlatformIO\Projects\Test_PortentaH7_M7core> C:\Users\jh\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.9.0-arduino2/dfu-util --device 0x2341:0x035b -D C:\Users\jh\Documents\PlatformIO\Projects\Test_PortentaH7_M7core\.pio\build\portenta_h7_m7\firmware.bin -a0 --dfuse-address=0x08040000:leave
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc. 
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY       
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/   

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Cannot open DFU device 2341:035b
Opening DFU capable USB device...
ID 2341:035b
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08040000, size = 114704
Download        [=========================] 100%       114704 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
PS C:\Users\jh\Documents\PlatformIO\Projects\Test_PortentaH7_M7core>
```

I will update the platform now.

datacode-taavi, 2021-04-01:
> Can someone try the latest platform revision from the dev branch?
> 
> ```ini
> [env:portenta_h7_m4]
> platform = https://github.com/platformio/platform-ststm32.git
> framework = arduino
> board = portenta_h7_m4
> ```

Unfortunately, no luck with dev branch (same result).

However, it seems that the issue is caused by the "-d" argument being sent to dfu-util.

I tried removing the other hwids from boards/portenta_h7_m7.json and now I can upload using pio **but only if I manually set the device to bootloader mode.** It still fails to do it by itself.

```
$ pio run -e portenta_h7m7 -t upload -v
Processing portenta_h7m7 (platform: https://github.com/platformio/platform-ststm32.git; board: portenta_h7_m7; framework: arduino; upload_protocol: dfu)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m7.html
PLATFORM: ST STM32 (12.1.0+sha.3d2553b) (git+https://github.com/platformio/platform-ststm32.git) > Arduino Portenta H7 (M7 core)
HARDWARE: STM32H747XIH6 480MHz, 511.35KB RAM, 768KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed 1.3.2 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library /home/taavi/.platformio/packages/framework-arduino-mbed/libraries/LittleVGL
Framework incompatible library /home/taavi/.platformio/packages/framework-arduino-mbed/libraries/Portenta_SDCARD
Framework incompatible library /home/taavi/.platformio/packages/framework-arduino-mbed/libraries/mbed-memory-status
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 24 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/portenta_h7m7/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.5% (used 55176 bytes from 523624 bytes)
Flash: [=         ]  14.5% (used 114072 bytes from 786432 bytes)
.pio/build/portenta_h7m7/firmware.elf  :
section               size        addr
.text               111640   134479872
.ARM.extab             520   134591512
.ARM.exidx            1032   134592032
.crash_data_ram        256   603980440
.data                 1400   603980696
.bss                 53776   603982096
.heap               467168   604035872
.openamp_section         0   939524096
.lwip_sec           278528   805306368
.ARM.attributes         44           0
.comment               126           0
.debug_info        2722120           0
.debug_abbrev       237710           0
.debug_aranges       26496           0
.debug_ranges        42584           0
.debug_line         372338           0
.debug_str          750637           0
.debug_frame         83932           0
.debug_loc          331401           0
.stab                   60           0
.stabstr               118           0
Total              5481886
<lambda>(["upload"], [".pio/build/portenta_h7m7/firmware.bin"])
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
[... 54 lines cut ...]
Download	[======                   ]  25%        28672 bytes
Download	[=======                  ]  28%        32768 bytes
Download	[========                 ]  32%        36864 bytes
Download	[=========                ]  39%        45056 bytes
Download	[==========               ]  42%        49152 bytes
Download	[===========              ]  46%        53248 bytes
Download	[============             ]  50%        57344 bytes
Download	[=============            ]  53%        61440 bytes
Download	[==============           ]  57%        65536 bytes
Download	[===============          ]  60%        69632 bytes
Download	[================         ]  64%        73728 bytes
Download	[=================        ]  71%        81920 bytes
Download	[==================       ]  75%        86016 bytes
Download	[===================      ]  78%        90112 bytes
Download	[====================     ]  82%        94208 bytes
Download	[=====================    ]  85%        98304 bytes
Download	[======================   ]  89%       102400 bytes
Download	[=======================  ]  92%       106496 bytes
Download	[======================== ]  96%       110592 bytes
Download	[=========================] 100%       114592 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
======================================================================================================== [SUCCESS] Took 4.33 seconds ========================================================================================================
```

maxgerhardt, 2021-04-01:
Good finding about the VID/PID pairs. Maybe dfu-util only takes the last one given to it? Or the Arduino supplied version is different.

>I can upload using pio but only if I manually set the device to bootloader mode

Mhm then there's maybe some fault in the logic that triggers the reset. Because one should e.g. see 

https://github.com/platformio/platform-ststm32/blob/3d2553b814e9d61758ee010986de6a3b3365dccc/builder/main.py#L222-L229

which doesn't appear in your verbose output.

><lambda>(["upload"], [".pio/build/portenta_h7m7/firmware.bin"])
AVAILABLE: cmsis-dap, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pio/build/portenta_h7m7/firmware.bin"])
Auto-detected: /dev/ttyACM0
"/home/taavi/.platformio/packages/tool-dfuutil/bin/dfu-util" -d 0x2341:0x035b -a 0 -s 0x08040000:leave -D "/home/taavi/CLionProjects/portenta_test/.pio/build/portenta_h7m7/firmware.bin"
dfu-util 0.9
>
>Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.

Also this doesn't seem to trigger

https://github.com/platformio/platform-ststm32/blob/3d2553b814e9d61758ee010986de6a3b3365dccc/builder/main.py#L38-L39

Hopefully @valeros can find out why that's the case.

valeros, 2021-04-01:
@datacode-taavi Thanks for the hints. Please pull the latest revision and try to upload again.

datacode-taavi, 2021-04-01:
> @datacode-taavi Thanks for the hints. Please pull the latest revision and try to upload again.

Now it works!

Thank you.

valeros, 2021-04-05:
Resolved in https://github.com/platformio/platform-ststm32/commit/e871689f812853855490e6919f5ed0a4aab9e12e

closed 2021-04-05 by valeros