platformio/platform-ststm32 #515: Deprecated STMF03C8 while referenced in MBED OS?

XavierGeerinck, 2021-04-05:
Hi,

I am quite new to all of this, so still catching up! :) 

I am trying to compile for the STMF103C8 with MBED OS 6.9.0 and saw that it was supported as shown here: https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.9.0/targets/TARGET_STM/TARGET_STM32F1/TARGET_STM32F103x8

However, when utilizing PlatformIO it refuses to start compiling for this target device. When looking in the code, I saw that it was under deprecated boards: https://github.com/platformio/platform-ststm32/blob/09c1cf74a98851358c950bdf74327b7385e1cd93/misc/mbed_deprecated_boards.json#L15

Question: Is this still deprecated then if the target is specified on MBED OS? I see STM32F103x8 though so am not sure if I am mixing them up

As far as I could find, the reason for this would be the small Flash Size? (64KB vs 128 required these days)

valeros, 2021-04-05:
> Is this still deprecated then if the target is specified on MBED OS? I see STM32F103x8 though so am not sure if I am mixing them up.

Yes, it's still deprecated. The only available F103-based target that is officially compatible with the latest Mbed releases is [`Nucleo F103RB`](https://os.mbed.com/platforms/ST-Nucleo-F103RB/).

closed 2021-04-05 by valeros

XavierGeerinck, 2021-04-05:
Ok! Thanks a lot for the fast response 😊