platformio/platform-ststm32 #518: Fix for STM32F7508-DK

jcw, 2021-04-08:
The build config for the STM32F7508-DK board appears to be incorrect.

* assert in `stm32cube.py` line 43, workaround: add `board_build.product_line = STM32F750xx` to my `platformio.ini` file
* it refuses to build with `framework = cmsis`, workaround: switch to `framework = stm32cube`

Both these issues can probably be solved with an update to `boards/disco_f750n8.json`.

valeros, 2021-04-08:
Hi @jcw ! Strange, but I cannot reproduce the issue, [disco_f750n8.json](https://github.com/platformio/platform-ststm32/blob/develop/boards/disco_f750n8.json#L7) already contains the `product_line` field. What is the version of your `ST STM32` platform?

jcw, 2021-04-08:
I see 12.1.1 when I do `pio update`.

jcw, 2021-04-08:
Very surprising. Both issues are gone here as well. I'll re-open if this comes back.
Thanks for checking!

closed 2021-04-08 by jcw

jcw, 2021-04-08:
Puzzle solved: this was in a demo project where I added my own boards file, probably before F750 was supported by PIO.