platformio/platform-ststm32 #521: STM32duino v2.0

sstaub, 2021-04-22:
Can you update STM32duino to v2.0?

ColdFireHunter, 2021-04-23:
If there is a need for help, I could make the .json Board Files, I did a lot of them recently by myself.

yqboy, 2021-04-23:
I just need it too, Can anyone tell me how to update ?

ColdFireHunter, 2021-04-23:
We need to transform all the variants from the core to the .json files for Platform I/O and then we need to update dependencies and version info.

yqboy, 2021-04-23:
I replaced framework-arduinoststm32 to V2.0 but it does not work. so it is the problem of .json file and dependencies?

fpistm, 2021-04-23:
@valeros if you need some information. Do not hesitate, I could help ;)

Main changes are in the variant management.
Now variant are defined per series/mcus. Ex S`TM32H7xx/H742Z(G-I)T_H743Z(G-I)T_H747A(G-I)I_H747I(G-I)T_H750ZBT_H753ZIT_H757AII_H757IIT`

* There is only one `variant.h` file which includes the correct `variant_<name>.h` file.
https://github.com/stm32duino/Arduino_Core_STM32/blob/2e9eeabc6c540e6157f83ec0ec08e29af47086e7/cores/arduino/variant.h#L15
So I guess PIO board json have to define the `VARIANT_H`
For all generic it is the same: `variant_generic.h`
For specific board it is `variant_<board_name>.h`

* By default all variant use the default `PeripheralPins.c` anyway it is possible to create a specific one for a board and use it by defining `CUSTOM_PERIPHERAL_PINS`. For example Daisy Seed define it to use its dedicated one:
https://github.com/stm32duino/Arduino_Core_STM32/blob/2e9eeabc6c540e6157f83ec0ec08e29af47086e7/boards.txt#L3531-L3540

https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants/STM32H7xx/H743I(G-I)(K-T)_H750IBK_H753II(K-T)

yqboy, 2021-04-23:
I don't know the operation of platfromIO, but I tried to modify the .json but the compilation error was reported. This is beyond my knowledge. I had to extract the V2.0 variants and add it to platfromIO, and it can now run well.

yqboy, 2021-04-23:
[image]
[image]

ColdFireHunter, 2021-04-23:
@fpistm Do you know how to include the new variant Files, because I tried it but couldn't get it to work? I already prepared my PIO with the new core v2.0 Update but couldn't get it to compile, I'm Missing the variant.h file. Thank you for your help.

fpistm, 2021-04-23:
Honestly I don't know so much about PIO. The only thing I do was to update the pio build GH action of the core. As mentioned in my previous comment.

https://github.com/stm32duino/Arduino_Core_STM32/blob/2e9eeabc6c540e6157f83ec0ec08e29af47086e7/.github/actions/pio-build/entrypoint.sh#L15-L21

ColdFireHunter, 2021-04-23:
Ok, then I will wait until valeros helps update the code. Thank you for your help and your work.

yqboy, 2021-04-23:
I tested it again and I found that I forgot to turn on ARDUINO_GENERIC_G030F6PX and it worked fine
[image]
Like fpistm said, the path does not support ‘(’and‘)’

yqboy, 2021-04-23:
> Honestly I don't know so much about PIO. The only thing I do was to update the pio build GH action of the core. As mentioned in my previous comment.
> 
> https://github.com/stm32duino/Arduino_Core_STM32/blob/2e9eeabc6c540e6157f83ec0ec08e29af47086e7/.github/actions/pio-build/entrypoint.sh#L15-L21

 I tested it again and I found that I forgot to turn on ARDUINO_GENERIC_G030F6PX and it worked fine
[image]
Like fpistm said, the path does not support ‘(’and‘)’
Thank you for your help

ColdFireHunter, 2021-04-24:
I now also got it to compile and run on multiple MCUs. But I have some formatting errors with missing whitespaces.

yqboy, 2021-04-24:
> I now also got it to compile and run on multiple MCUs. But I have some formatting errors with missing whitespaces.

I tested F030F4 F103C8 G030F6/K6 MCUs. everything is ok
[image]

ColdFireHunter, 2021-04-25:
I now also have no errors. I have now updated all the files needed for  F0/F1/F2/F4 and I'm working on F7/G0. When I am finished I will open a Pull Request. Have a nice day.

valeros, 2021-05-18:
Hi guys, I've updated the Arduino package to the latest v2.0, please retest your projects with the upstream version:
```ini
[env:test]
platform = https://github.com/platformio/platform-ststm32.git
framework = arduino
board = ...
```

fpistm, 2021-05-21:
Hi @valeros 
since your latest changes our pio CI is failed with:

```console
 Error: Could not load broken JSON: /github/home/.platformio/platforms/ststm32/boards/disco_b_g431b_esc1.json
```

Full log [here](https://github.com/stm32duino/Arduino_Core_STM32/pull/1253/checks?check_run_id=2631687623#step:3:755)

valeros, 2021-05-24:
Hi @fpistm ! Thanks for reporting. Should be fixed.

fpistm, 2021-05-24:
> 
> 
> Hi @fpistm ! Thanks for reporting. Should be fixed.

Hi @valeros 

Thankk you.

Go further but always failed:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1402/checks?check_run_id=2655959954#step:3:790
```console
 Error: Missing PlatformIO build script /github/home/.platformio/packages/framework-arduinoststm32/tools/platformio/platformio-build.py!
```

valeros, 2021-05-24:
Thanks, I submitted a PR with the latest changes.

jlocsx, 2021-06-02:
Hi @valeros,

Even upgrading to ST STM32 [v14.0.0](https://github.com/platformio/platform-ststm32/releases/tag/v14.0.0) some files of the STM32 Arduino Core remain in  [v1.9.0](https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.9.0) instead of [v2.0.0](https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/2.0.0).

For example, the `USBSerial.cpp` and `USBSerial.h` files (\~/platformio/packages/framework-arduinoststm32/cores/arduino/) still remain in STM32 Arduino Core v1.9.0, but `stm32_def.h` file (\~/platformio/packages/framework-arduinoststm32/cores/arduino/stm32/) has been updated.

I noticed this issue when trying to use this new API: https://github.com/stm32duino/Arduino_Core_STM32/pull/1382

valeros, 2021-06-02:
Hi @jlocsx ! That's strange, we use the same package from https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json . Are you able to use that new API from the Arduino IDE?

fpistm, 2021-06-02:
In fact there is no issue. The new API's you are referring  [here](https://github.com/stm32duino/Arduino_Core_STM32/pull/1382)
are not part of the 2.0.0. They are merged in the master after the release and so will be part of the next release 2.x.x as stated in the milestone.

Ralf9, 2021-07-10:
Hi,
with the core v2.0 and the Bootloader 2.0 I get at build a fatal error. I use Windows 10 and Visual Studio Code
```
Processing maple_mini_b20 (platform: https://github.com/platformio/platform-ststm32.git; board: maple_mini_b20; framework: arduino)
--------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/maple_mini_b20.html
PLATFORM: ST STM32 (14.0.1+sha.f702b3f) > Maple Mini Bootloader 2.0        
HARDWARE: STM32F103CBT6 72MHz, 20KB RAM, 120KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.20000.210603 (2.0.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - tool-dfuutil 1.9.200310
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Warning! Cannot find linker script for the current target!

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o
<command-line>: fatal error: variant_MAPLE_MINI_B20.h: No such file or directory
```
here is the platformio.ini
```
[env:maple_mini_b20]
platform = https://github.com/platformio/platform-ststm32.git
board = maple_mini_b20
board_build.core = stm32
framework = arduino
```
and here the main.cpp
```
#include <Arduino.h>

#ifndef LED_BUILTIN
  #define LED_BUILTIN 33
#endif

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

Ralf9, 2021-07-11:
when I make a build with verbose this is the command before the fatal error:
```
arm-none-eabi-gcc -o .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o -c -std=gnu11 -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=50101 -DSTM32F103xB -DSTM32F1xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_MAPLE_MINI_B20 -DBOARD_NAME=\"MAPLE_MINI_B20\" -DHAL_UART_MODULE_ENABLED -DUSE_FULL_LL_DRIVER -DVARIANT_H=\"variant_MAPLE_MINI_B20.h\" -DBL_LEGACY_LEAF -DVECT_TAB_OFFSET=0x2000 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\avr 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\OpenAMP 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Inc 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Src 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\STM32F1xx 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Src 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\open-amp\lib\include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\libmetal\lib\include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\virtual_driver 
-IC:\Users\Ralf\.platformio\packages\framework-cmsis\CMSIS\Core\Include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Source\Templates\gcc 
-IC:\Users\Ralf\.platformio\packages\framework-cmsis\CMSIS\DSP\Include 
-IC:\Users\Ralf\.platformio\packages\framework-cmsis\CMSIS\DSP\PrivateInclude 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\variants\MAPLEMINI_F103CB C:\Users\Ralf\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_can.c
```
The problem is, in the core 2.0. the variants are in:
`variants\STM32F1xx\F103C8T_F103CB(T-U)`
and there is no "variant_MAPLE_MINI_B20.h"

I also noticed that at release 13 and 14 the arduino- and platformio version is the same. Is this ok?
`ARDUINO=10808 PLATFORMIO=50101`

In the arduino IDE with core 2.0 is
`ARDUINO=10815 `

valeros, 2021-11-02:
Resolved in [v14.0](https://github.com/platformio/platform-ststm32/releases/tag/v14.0.0)

closed 2021-11-02 by valeros

Ralf9, 2021-11-03:
With v14.0, v14.2 and v15 I have  still this error
`<command-line>: fatal error: variant_MAPLE_MINI_B20.h: No such file or directory`

cennox-ashish, 2023-03-06:
Hello,

I am working on a project which involves STM32WB5MMG. but platform.io doesn't have supporting board. 
and I was trying to follow the steps mentioned in this thread. but I am facing issues.

here is my .json file
`{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32WB5Mxx -DSTM32WBx DUSE_HAL_DRIVER",
    "f_cpu": "64000000L",
    "mcu": "stm32wb5mmgh",
    "product_line": "STM32WB5Mxx",
    "variant": "STM32WBxx/WB5MMGH"
  },
  "connectivity": [
    "bluetooth"
  ],
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32WB5Mxx",
    "openocd_target": "stm32wbx",
    "svd_path": "STM32WB55_CM4.svd"
  },
  "frameworks": [
    "arduino",
    "mbed",
    "stm32cube"
  ],
  "name": "STM32WB5MMG",
  "upload": {
    "maximum_ram_size": 262144,
    "maximum_size": 1048576,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "blackmagic"
    ]
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32wb5mmg.html",
  "vendor": "ST"
}
`
and platformio.ini

`[env:STM32WB5MMG]
platform = ststm32
board = STM32WB5MMG
framework = arduino
upload_protocol= stlink`

and this is the error I'm getting

Processing STM32WB5MMG (platform: ststm32; board: STM32WB5MMG; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32WB5MMG.html
PLATFORM: ST STM32 (15.5.1) > STM32WB5MMG
HARDWARE: STM32WB5MMGH 64MHz, 256KB RAM, 1MB Flash
DEBUG: Current (stlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.20400.0 (2.4.0)
 - framework-cmsis @ 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
[... 98 lines cut ...]
*
*****************************************************************************

compilation terminated.
compilation terminated.
*** [.pio\build\STM32WB5MMG\SrcWrapper\src\HAL\stm32yyxx_hal_crc.c.o] Error 1
*** [.pio\build\STM32WB5MMG\SrcWrapper\src\HAL\stm32yyxx_hal_cortex.c.o] Error 1
<command-line>: fatal error: variant_STM32WB5MMG.h: No such file or directory

*****************************************************************************
* Looking for variant_STM32WB5MMG.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:variant_STM32WB5MMG.h"
* Web  > https://registry.platformio.org/search?q=header:variant_STM32WB5MMG.h
*
*****************************************************************************

compilation terminated.
*** [.pio\build\STM32WB5MMG\SrcWrapper\src\HAL\stm32yyxx_hal_crc_ex.c.o] Error 1
=================================================================== [FAILED] Took 1.71 seconds ===================================================================

 *  The terminal process "C:\Users\ashish.jaiswal\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

I would appreciate if anyone could help and teach me how to resolve this issue. I am pretty new in stm enviroment.

mkeyno, 2023-07-30:
hi @all

I've tried to use this solution for my project using the STM32G030P6 board within platformio  with Arduino platform 
since there is no predefined STM32G030P6  board in Platformio, I created my own board with the following parameters 

```json

{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m0plus",
    "extra_flags": "-DSTM32G0xx -DSTM32G030xx -DVARIANT_H=\\\"variant_generic.h\\\"",
    "f_cpu": "64000000L",
    "framework_extra_flags": {
      "arduino": "-D__CORTEX_SC=0"
    },
    "mcu": "stm32g030f6p6",
    "product_line": "STM32G030xx",
    "variant": "STM32G0xx/G030F6P"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32G030F6",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_target": "stm32g0x",
    "svd_path": "STM32G030.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "stm32cube"
  ],
  "name": "Generic STM32G030F6",
  "upload": {
    "maximum_ram_size": 8192,
    "maximum_size": 32768,
    "protocol": "stlink",
    "protocols": [
      "blackmagic",
      "cmsis-dap",
      "dfu",
      "jlink",
      "serial",
      "stlink",
      "mbed"
    ]
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32g030f6.html",
  "vendor": "ST"
}
```

however when I create the new project with the Arduino framework, the `main.c` default code could not compile due to a reference error
the error is located in the [“%USER%.platformio\packages\framework-arduinoststm32\cores\arduino\variant.h” ](https://github.com/stm32duino/Arduino_Core_STM32/blob/2e9eeabc6c540e6157f83ec0ec08e29af47086e7/cores/arduino/variant.h)path
it can not find any file with the name of ` #include "variant_STM32G030F6P6.h"`

then I change the code to this  `#include "variant_generic.h" `

however, when I compile the code, it throws considerable errors to the definition of Arduino’s basic function

```
Compiling .pio\build\stm32g030f6p6\SrcWrapper\src\HAL\stm32yyxx_hal.c.o
<command-line>: warning: "VARIANT_H" redefined
<command-line>: warning: "VARIANT_H" redefined
<command-line>: note: this is the location of the previous definition
<command-line>: note: this is the location of the previous definition
[... 751 lines cut ...]
hw_config.c:(.text.hw_config_init+0xe): undefined reference to `SystemClock_Config'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/stm32g030f6p6/libFrameworkArduino.a(HardwareSerial.cpp.o): in function `HardwareSerial::setRx(unsigned long)':
HardwareSerial.cpp:(.text._ZN14HardwareSerial5setRxEm+0x48): undefined reference to `digitalPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: HardwareSerial.cpp:(.text._ZN14HardwareSerial5setRxEm+0x4c): undefined reference to `analogInputPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/stm32g030f6p6/libFrameworkArduino.a(HardwareSerial.cpp.o): in function `HardwareSerial::setTx(unsigned long)':
HardwareSerial.cpp:(.text._ZN14HardwareSerial5setTxEm+0x48): undefined reference to `digitalPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: HardwareSerial.cpp:(.text._ZN14HardwareSerial5setTxEm+0x4c): undefined reference to `analogInputPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/stm32g030f6p6/libFrameworkArduino.a(HardwareSerial.cpp.o): in function `HardwareSerial::HardwareSerial(void*, HalfDuplexMode_t)':
HardwareSerial.cpp:(.text._ZN14HardwareSerialC2EPv16HalfDuplexMode_t+0x90): undefined reference to `analogInputPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: HardwareSerial.cpp:(.text._ZN14HardwareSerialC2EPv16HalfDuplexMode_t+0x98): undefined reference to `digitalPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/stm32g030f6p6/libFrameworkArduino.a(wiring_digital.c.o): in function `pinMode':
wiring_digital.c:(.text.pinMode+0x9c): undefined reference to `digitalPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.pinMode+0xa8): undefined reference to `analogInputPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/stm32g030f6p6/libFrameworkArduino.a(wiring_digital.c.o): in function `digitalWrite':
wiring_digital.c:(.text.digitalWrite+0x68): undefined reference to `digitalPin'
c:/users/lkeyn/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: wiring_digital.c:(.text.digitalWrite+0x74): undefined reference to `analogInputPin'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\stm32g030f6p6\firmware.elf] Error 1
========================================================================================== [FAILED] Took 95.77 seconds ==========================================================================================

 *  The terminal process "C:\Users\lkeyn\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

any help is really appreciated