platformio/platform-ststm32 #525: disco_f303vc: arduino framework nt supported error

insomnimus, 2021-05-24:
I've got an stm32-f3 discovery board.

I wanted to use the arduino framework for a small project, however it seems that the board is not supported.

I've searched around and there are arduino cores for it, see [this page](https://github.com/stm32duino/Arduino_Core_STM32).

Is there a workaround I can follow to use the Arduino Framework with my stm32-f3 discovery board?

svenwiltink, 2021-05-24:
Looks like it requires arduino core 2.0.0, which is being worked on in https://github.com/platformio/platform-ststm32/issues/521

insomnimus, 2021-05-27:
Closing since there's already a more active discussion as you linked and i don't want to dupe issues.

closed 2021-05-27 by insomnimus