platformio/platform-ststm32 #528: In the v14.0.0 version, the serial port cannot work

yulong88888, 2021-06-01:
First of all thanks to all contributors for making it easy to use stm32.
I tried to use version 14 today, the serial port does not work, but there is no problem with version 13. The hardware I use is https://github.com/WeActTC/BluePill-Plus.
Hope you can get help, thank you very much.

[image]

This is my code

```
#include <Arduino.h>

void setup() {
   pinMode(PB_2, OUTPUT);

   Serial.begin(115200);
}

void loop() {
   digitalWrite(PB_2, HIGH);
   Serial.print("0.0");
   delay(1000);
   digitalWrite(PB_2, LOW);
   Serial.print("1.1");
   delay(1000);
}
```

valeros, 2021-06-01:
Hi @yulong88888 ! Thanks for reporting. Does it work in Arduino IDE?

yulong88888, 2021-06-01:
I have not used it in Arduino IDE, only used it in PIO

valeros, 2021-06-01:
It would be great if you could try to compile and upload your example using Arduino IDE, just to make sure there is no regression in the framework itself.

yulong88888, 2021-06-01:
I tried the Arduino IDE first, but when there are multiple cpp files to compile, it becomes difficult.   : )

fpistm, 2021-06-01:
Hi @yulong88888 

Your sketch is not correct. `PB_2` should be replaced by `PB2` because it does not represent the same thing. PB_2 is the PinName used in low level while PB2 is the pin number used by Arduino API.

About the Serial you means the Hardware Serial or the USB in CDC mode ?
I know both works with Arduino but never tested this BP plus. Moreover it seems there is 2 version on with a STM32F103 and one with a GD32F303CCT6, which one you used ?

yulong88888, 2021-06-01:
I changed "PB_2" to "PB2", the USB serial port still cannot be used, I use ch340, not cdc, and the hardware is sure to be stm32f103.

This is the new code
```
#include <Arduino.h>

void setup() {
   pinMode(PB2, OUTPUT);
   Serial.begin(115200);
}

void loop() {
   digitalWrite(PB2, HIGH);
   Serial.print("0.0");
   delay(1000);
   digitalWrite(PB2, LOW);
   Serial.print("1.1");
   delay(1000);
}
```

fpistm, 2021-06-01:
I did not expect the Serial works changing to PB2.
So you should have connected your ch340  on pins PA10/PA9 (Rx/Tx).
I have no idea why it does not work as told in my previous post BP has been tested and works as expected with Arduino IDE.
Maybe a mistake in the PIO integration/definition?

fpistm, 2021-06-01:
Hi @valeros
How is defined the `ARDUINO_<board_name>` in PIO ?

In this case `ARDUINO_BLUEPILL_F103C8` should be defined.

valeros, 2021-06-01:
@yulong88888 Please attach here your `platformio.ini` file.

yulong88888, 2021-06-01:
Sorry, I just got home, so I am late to reply.
I have been using stm32duino for more than a year. I upgraded to 14.0.0 today. I found that the previous project could not run anymore (this project mainly uses serial port to connect with ROS), so I used pio home to create a new project without changing the ini file .

```
; PlatformIO Project Configuration File
;
; Build options: build flags, source filter
; Upload options: custom upload port, speed and extra flags
; Library options: dependencies, extra library storages
; Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
```

valeros, 2021-06-01:
@yulong88888 I don't see any configuration flags in your env. Have you seen [this page](https://docs.platformio.org/en/latest/platforms/ststm32.html#id5) ? I tried the following config and it seems to work just fine:
```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
build_flags = 
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
```
```bash
 λ pio device monitor -p COM43 -b 115200
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM43  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
0.01.10.01.10.01.10.01.1
--- exit ---
```

yulong88888, 2021-06-01:
Sorry, I am using ch340 and pins PA10/PA9 (Rx/Tx), not CDC

yulong88888, 2021-06-01:
In the case that I did not change the hardware, the 13 version can work normally, but the 14 does not work. And when testing, only keep one of them.

fpistm, 2021-06-01:
In fact generic uses pins `PA3/PA2` while the dedicated variant for BP uses `PA10/PA9`.
So I guess you should set `bluepill_f103c8 `instead of `genericSTM32F103C8` if you want `PA10/PA9`

yulong88888, 2021-06-01:
I can try it tomorrow, but I still don't understand why the same program can't run on 14. :)

valeros, 2021-06-01:
> I can try it tomorrow, but I still don't understand why the same program can't run on 14. :)

The default serial pins for Generic F103 boards were changed in the latest Arduino core v2.0. I propose two options:
- Switch to the `bluepill_f103c8` which uses PA9/PA10 by default
- Override serial pins according to your needs
```ini
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
build_flags = 
    -DPIN_SERIAL_RX=PA10
    -DPIN_SERIAL_TX=PA9

```

closed 2021-06-01 by valeros

yulong88888, 2021-06-01:
Thank you very much

yulong88888, 2021-06-08:
Sorry, due to the short period of time, last time I changed the board from genericSTM32F103C8 to bluepill_f103c8, which temporarily solved the serial port problem.
I tried it today and I think the best way is this
```
# platform.ini
build_flags =
     -D ENABLE_HWSERIAL1

# main.cpp
# !!! The serial number is very important
Serial1.begin(115200);
Serial1.println("hello world");
```