platformio/platform-ststm32 #529: STM32 V14, Does not create new projects, Generates error when compiling old projects.

Merdock1, 2021-06-03:
this error recurs when trying to compile old projects.
If I change to version 13, everything works fine..

`Processing blackpill_f411ce (platform: ststm32; board: blackpill_f411ce; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/blackpill_f411ce.html
PLATFORM: ST STM32 (14.0.0) > WeAct Studio BlackPill V2.0 (STM32F411CE)
HARDWARE: STM32F411CEU6 100MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.20000.210601 (2.0.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
ImportError: cannot import name 'IS_WINDOWS' from 'platformio.compat' (%USERPATH%\.platformio\penv\lib\site-packages\platformio\compat.py):
  File "%USERPATH%\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "%USERPATH%\.platformio\platforms\ststm32\builder\main.py", line 120:
    target_elf = env.BuildProgram()
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "%USERPATH%\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "%USERPATH%\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "%USERPATH%\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "%USERPATH%\.platformio\platforms\ststm32\builder\frameworks\arduino.py", line 50:
    SConscript(build_script)
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 654:
    return method(*args, **kw)
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "%USERPATH%\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "%USERPATH%\.platformio\packages\framework-arduinoststm32\tools\platformio\platformio-build.py", line 28:
    from platformio.compat import IS_WINDOWS`

valeros, 2021-06-03:
Hi @Merdock1 ! It seems that you're using an outdate Arduino package,  could you please run `pio platform update ststm32`?

valeros, 2021-07-27:
Feel free to reopen the issue if you still need any help.

closed 2021-07-27 by valeros

Merdock1, 2021-07-29:
> Feel free to reopen the issue if you still need any help.

Hi @valeros .
Hello, sorry for the delay in replying, but the notice email was lost.
Fix the problem by compiling with version ststm32@^13.0.0
later I did some test with the version ststm32@^14.0.1
compiles without errors but MCU apparently not working.
Then the project was working with the version ststm32@^13.0.0.

Thanks for your help and my apologies again for the delay in replying.

valeros, 2021-07-29:
Does it work in Arduino IDE with the latest stable core v2.0?

Merdock1, 2021-07-29:
I did not test the Arduino Ide, the project is built in Platformio VSCode

valeros, 2021-07-30:
Would be great if you could run and upload a simple sketch using Arduino IDE just to verify there is no problem with the framework.

Merdock1, 2021-07-31:
> Would be great if you could run and upload a simple sketch using Arduino IDE just to verify there is no problem with the framework.

Hello, yesterday Platformio updated to Version 4.1
On Monday I will do some tests and inform you of the results.
Thanks for your time.
Greetings and have a good weekend.