platformio/platform-ststm32 #530: STM32 Compilation error

JulioCesarMatias, 2021-06-04:
Hi,

I'm having a compilation problem with the STM32F103, you updated the core and caused the Arduino.h extension to return *h(variant_PILL_F103Cx.h) error. I've tried the `pio update` and `pio upgrade --dev` commands, but it didn't work.PlatformIO Core `version 5.2.0a6`

Platformio.ini >> https://github.com/JCFLIGHT/JCFLIGHT/blob/master/platformio.ini

Error after compilation:

arm-none-eabi-g++: error: C:/Users/Julio: No such file or directory
arm-none-eabi-g++: error: Cesar/.platformio/packages/framework-arduinoststm32/variants/STM32F1xx/F103C8T_F103CB(T-U)/ldscript.ld: No such file or directory
arm-none-eabi-g++: error: Cesar/.platformio/packages/framework-arduinoststm32/variants/STM32F1xx/F103C8T_F103CB(T-U): No such file or directory
*** [.pio\build\JCFlightPascal\firmware.elf] Error 1

[image]

valeros, 2021-06-04:
Hi @JulioCesarMatias ! Could you please try to update the platform package to the latest version? Just run `pio platform update ststm32` and try again.

JulioCesarMatias, 2021-06-04:
didn't solve

[image]

JulioCesarMatias, 2021-06-04:
[image]

valeros, 2021-06-04:
Please run `pio platform show ststm32` in the IDE terminal and upload the output to https://pastebin.com/

JulioCesarMatias, 2021-06-04:
> Please run `pio platform show ststm32` in the IDE terminal and upload the output to https://pastebin.com/

https://pastebin.com/hWHVDYe7

valeros, 2021-06-04:
Thanks, also could you please clean the project, compile it in verbose mode and upload the log to https://pastebin.com/ as well:
[image]

JulioCesarMatias, 2021-06-04:
> Obrigado, você também poderia limpar o projeto, compilá-lo em modo detalhado e enviar o log para https://pastebin.com/ também:
> [image]

https://pastebin.com/eb79Wfuh

valeros, 2021-06-04:
It seems that you didn't copy the entire log, I can only see few lines. You need to clean your project and only then compile it in verbose mode.

JulioCesarMatias, 2021-06-04:
> It seems that you didn't copy the entire log, I can only see few lines. You need to clean your project and only then compile it in verbose mode.

I updated, I put the entire log

valeros, 2021-06-04:
Strange, but I cannot reproduce the issue. Could you try to delete the `C:\Users\Julio Cesar\.platformio` folder and try again?

JulioCesarMatias, 2021-06-04:
After I did the Verbose Build, the problem disappeared, now it compiles normally. This was a PlatformIO Bug, maybe it is not cleaning up what is needed for a new build. I as a programmer found it strange too, but @valeros thanks for your attention!

closed 2021-06-04 by JulioCesarMatias