platformio/platform-ststm32 #537: Code appears to upload but does not run on STM32 Nucleo L476RG

nikob997, 2021-06-16:
I recently updated VScode on my mac and since then any code upload onto my Nucleo L476RG appears to upload (the indicator light flashes red and green when uploading), but the code doesn't run; anything I try to output to the serial port doesn't show up, and a test code I created to blink the built-in LED doesn't work either. I've tried changing the upload protocol, reverting back to the old version of VScode, even reverting back to an older version of the C++ extension on VScode, but nothing works. My friend can run everything exactly the same on a PC with no issues, which makes me think this is a mac problem; I'm just not sure what went wrong. 

Test code:

```
// Start program

#include <Arduino.h>
#include <PrintStream.h>

void setup() 
{
    pinMode(LED_BUILTIN,OUTPUT);
    digitalWrite(LED_BUILTIN,HIGH);

    // Open serial port
    Serial.begin(115200);
    //Wait for connection to be set up
    delay(1000);

    // Intro Output
    uint16_t class_num = 507;
    Serial << endl << "Starting Program";
    Serial << endl << "Class: ME " << class_num << endl;

    pinMode(LED_BUILTIN,OUTPUT);
    digitalWrite(LED_BUILTIN,HIGH);
}

void loop() 
{
    Serial.print(" - "); 

    digitalWrite(LED_BUILTIN,HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN,LOW);
    delay(1000);
}
```

Platformio.ini: 

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:nucleo_l476rg]
platform = ststm32
board = nucleo_l476rg
framework = arduino

;bits per second from mC to computer to make sure they are the same frequency- otherwise file downloads will not work
monitor_speed = 115200

upload_protocol = stlink
```

gudnimg, 2021-06-16:
* What version of Mac OS are you using?
* Version of `ststm32`? (14.0.0?)
* Version of PIO?
* Version of VScode?
* Version of `stm32duino`? (used for Arduino framework as I understand it)

Do you have another Nucleo board to test if it works?

nikob997, 2021-06-16:
Versions: 
Mac: macOS Catalina 10.15.7
ststm32: 14.0.0
PlatformIO IDE: 2.3.2 (tried 2.3.1 also)
VSCode: 1.57.0 (tried 1.56.0 also)
stm32duino: most current? I can't find where I can see the version of this
C/C++ VScode extension: 1.4.1

I have tried using a second board, but I get the same results. Both boards work fine when I test the same code on my friends PC, and also still run if I compile and upload the code from the PC and then just connect the board to my mac without re-uploading the code. I've pretty much ruled it down to being some kind of issue with the code upload process on my mac, since that is the only thing that causes the error. Even if I have working code running on the board installed from the PC, if I re-upload it from my mac, it will wipe the old code and install the code again (as it should), which unfortunately stops the boards from working.

nikob997, 2021-06-16:
Here's the full output to the terminal when I upload the code- it looks like it has some versions of things at the beginning, under "Packages". It says it uploads successfully, but the code does not actually run on the microcontroller.

```
Processing nucleo_l476rg (platform: ststm32; board: nucleo_l476rg; framework: arduino)
-----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_l476rg.html
PLATFORM: ST STM32 (14.0.0) > ST Nucleo L476RG
HARDWARE: STM32L476RGT6 80MHz, 128KB RAM, 1MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-arduinoststm32 4.20000.210603 (2.0.0) 
 - framework-cmsis 2.50700.210515 (5.7.0) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1100.0 (11.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PrintStream> 0.1.0+sha.6a9e0d3
|-- <STM32duino FreeRTOS> 10.2.1+sha.d624467
Building in release mode
Checking size .pio/build/nucleo_l476rg/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.9% (used 1176 bytes from 131072 bytes)
Flash: [          ]   1.9% (used 19856 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/nucleo_l476rg/firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.11.0-00155-ge392e485e (2021-03-15-18:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08003e80 msp: 0x20020000
** Programming Started **
Warn : Adding extra erase range, 0x08004f40 .. 0x08004fff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
========================================== [SUCCESS] Took 5.90 seconds ==========================================

Terminal will be reused by tasks, press any key to close it.
```

gudnimg, 2021-06-17:
I get this error when I build your code:

[image]

Is `PrintStream.h` something you keep locally in your project?

gudnimg, 2021-06-17:
On my end the version of stm32duino is 1.0.2, not sure if it is critical.

akasona, 2021-06-18:
I have similar problem.
And it's my simple steps to reproduce and workaround below:

1. newly create project via PlatfprmIO IDE. My platformio.ini is:
    ```
    [env:nucleo_l476rg]`
    platform = ststm32
    board = nucleo_l476rg
    framework = arduino
    ```
2. begin debugging, with default empty setup() and loop().
3. EXPECTED: the program automatically breaks near main().
RESULT: the program keeps running in Default_Handler (startup_stm32l476xx.s).
4. WORKAROUND: downgrade ststm32 to 13.0.0. My platformio.ini is:
    ```
    [env:nucleo_l476rg]`
    platform = ststm32@13.0.0
    board = nucleo_l476rg
    framework = arduino
    ```

Versions:
Windows 10 Pro
ststm32: 14.0.0
PlatformIO IDE: 2.3.2
VSCode: 1.57.0
stm32duino: most current? I can't find where I can see the version of this
C/C++ VScode extension: 1.4.1

steph643, 2021-06-18:
Same issue here.
@akasona's workaround seems to work well.
There's an (unconclusive) discussion about this issue [here](https://community.platformio.org/t/nucleo-board-upload-issue/21943/2).

maxgerhardt, 2021-06-18:
Ohh I possibly know why it all broke. 

You see the LD script

https://github.com/stm32duino/Arduino_Core_STM32/blob/03052da8712de2e2f48b0460da079ff3c50e3fb0/variants/STM32L4xx/L475R(C-E-G)T_L476R(C-E-G)T_L486RGT/ldscript.ld#L37-L43

has two RAM regions. PlatformIO substitutes for `LD_MAX_DATA_SIZE` the value of the *whole RAM*, 

https://github.com/platformio/platform-ststm32/blob/62cae8a55ccb48ed06bc2edc87ca25f9350399c9/boards/nucleo_l476rg.json#L36-L36

and not just the length of the first RAM bank (which is 96KB and not 128KB).

Thus the stack pointer is intialized wrongly in a section of RAM that does not exist

https://github.com/stm32duino/Arduino_Core_STM32/blob/03052da8712de2e2f48b0460da079ff3c50e3fb0/variants/STM32L4xx/L475R(C-E-G)T_L476R(C-E-G)T_L486RGT/ldscript.ld#L32-L32

and this instantly crashes on anything that needs stack, e.g. pushing registers to the stack.

Does it magically work if you add

```ini
build_upload.maximum_ram_size = 98304
```
at the end of the `platformio.ini`?

steph643, 2021-06-18:
>Does it magically work if you add
> `build_upload.maximum_ram_size = 98304`
>at the end of the platformio.ini when using the 14.0.0 platform?

No, it doesn't work.

akasona, 2021-06-18:
@maxgerhardt I think "build_upload.maximum_ram_size" has a typo or something wrong.
There's an error message in the TERMINAL:

```
Warning! Ignore unknown configuration option `build_upload.maximum_ram_size` in section [env:nucleo_l476rg]
```

steph643, 2021-06-18:
Indeed, good catch @akasona, same for me.

maxgerhardt, 2021-06-18:
Sorry I made a mistake, 

```ini
board_upload.maximum_ram_size = 98304
```

steph643, 2021-06-18:
Yes, it seems to solve the issue.

akasona, 2021-06-18:
It works fine!!!!!

```
[env:nucleo_l476rg]
platform = ststm32
board = nucleo_l476rg
framework = arduino
board_upload.maximum_ram_size = 98304
```

maxgerhardt, 2021-06-18:
Alright then we *absolutely* have a problem in the PlatformIO builder script and board information for dual RAM-bank boards.

nikob997, 2021-06-18:
> I get this error when I build your code:
> 
> [image]
> 
> Is `PrintStream.h` something you keep locally in your project?
> 
> Edit:
> 
> I see this in your comment above:
> 
> ```
> |-- <PrintStream> 0.1.0+sha.6a9e0d3
> |-- <STM32duino FreeRTOS> 10.2.1+sha.d624467
> ```
> 
> did you not post the full `platformio.ini` file perhaps?

Ah yeah, my bad it looks like I missed the last couple lines; printstream is another library I've been using to print to the serial port, which has always worked fine in the past. Add these lines to the `platformio.ini` file to add that in:

```
lib_deps =
   https://github.com/tttapa/Arduino-PrintStream.git
   https://github.com/stm32duino/STM32FreeRTOS.git
```

The test code I made doesn't need STM32FreeRTOS to work, so you can probably leave that line out.

nikob997, 2021-06-18:
Thank you so much everyone! Adding the line `board_upload.maximum_ram_size = 98304` seems to fix the problem.

microtronics, 2021-06-20:
Yep, this issue really puzzled me. Debugger immediately jumps into the FaultHandler and it took me quite a while to find out that it happened when I updated from 13.0.0 to 14.0.0...

valeros, 2021-06-21:
Thanks for pointing this out, should be fixed in the dev branch.

closed 2021-06-21 by valeros

maxgerhardt, 2021-07-14:
@valeros Certain F4 boards with Core-Coupled memory have the exact same problem, see https://community.platformio.org/t/stm32-nucleo-arduino-pin-definitions-wrong/22470/5?u=maxgerhardt

maxgerhardt, 2021-07-14:
On a general note, there has to be a better solution there. The fix commit just reduces the RAM size that PlatformIO thinks that board has. But the board has more, just in different locations. There should a be a new entry `ram0_size` or something where it specifies the length of continuous RAM located at the standard 0x2000000, but in the final RAM size check still use the whole RAM length. Otherwise some special firmwares which make use of the other SRAM banks will fail this check due to 'not enough memory' when there actually is.

microtronics, 2021-07-14:
@maxgerhardt I had a similar issue 2 years ago with F303K8. Part of the problem is the linker script. The ones generated by STM32Cube do include the ccram section, but without explicitly telling the linker to use it, it won't be used by default. This leads to the memory gauge showing around 70% free ram and when you add a small array, it's suddenly exceeded.
I guess the ram size check and memory gauge should be extended to support different regions in a more generic way...

maxgerhardt, 2021-07-17:
And another one at https://community.platformio.org/t/code-not-executing-on-stm32f3-discovery-with-arduino-framework/22509 having the same problem with `disco_f303vc`

https://github.com/platformio/platform-ststm32/blob/c55a3759bb40beb783ddb38fa3a325c5acca0861/boards/disco_f303vc.json#L36-L37 

The board has 40kByte SRAM1 (0x2000_0000) and 8kByte Core-Coupled-Memory (0x1000_0000).

commit 38e325b5 (2021-06-21): Sync RAM sizes for Arduino-enabled L4-based boards // Resolve #537

STM32L4 targets may have several separate blocks of SRAM,
for example STM32L476RGT6 has two blocks (96KB + 32KB)
To properly configure Arduino framework we need pass the size of
the first bank SRAM1 via the LD_MAX_DATA_SIZE define.