platformio/platform-ststm32 #539: please remove "stm32f7xx_hal_conf.h" from stm32cubef7

jcw, 2021-06-26:
I'm using code from a STM32 library which is part of `STM32CubeF7`:

    .platformio/packages/framework-stm32cubef7/Drivers/STM32F7xx_HAL_Driver/

Everything works perfectly, except that the file `STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_conf.h` prevents configuring the code for my setup. This file is _identical_ to `stm32f7xx_hal_conf_template.h`, in that same directory.

AIUI, the idea is to copy that template file to your own project, adjust as needed, and make sure it's in the include-path. This is how this project works, for example (makefile, not PIO): https://github.com/MayaPosch/FreeRTOS_HTTP_server

If I remove `stm32f7xx_hal_conf.h` from PIO's version, things work perfectly.

Such a configuration header does not belong _inside_ the library, please consider removing it.

jcw, 2021-06-27:
Hmmm ... is this file constantly being re-created by PlatformIO?
How can I override its defaults?

I'm using a Nucleo-F767ZI, which has a different Ethernet PHY, and get this error:

```
src/ethernetif.c: In function 'low_level_init':
src/ethernetif.c:164:31: error: 'LAN8742A_PHY_ADDRESS' undeclared (first use in this function); did you mean 'DP83848_PHY_ADDRESS'?
```

valeros, 2021-06-29:
Hi @jcw ! 

There is an undocumented setting for such cases, could you please try adding the following line to your configuration:
```ini
platform = ststm32
framework = stm32cube
board = ...
board_build.stm32cube.custom_config_header
```

jcw, 2021-06-29:
Yeay, that does exactly what I was after! \o/

All I can say is: *please* document this :)

Hawxxer, 2021-10-31:
Yes please document this somewhere, searched a whole day, why, with the code copied directly from stm32cube ide, my clock speeds do not get set correctly. Adding the setting made it work...

closed 2021-11-02 by valeros