platformio/platform-ststm32 #542: Unable to compile empty Nucleo H743 project

kyleb-imp, 2021-07-08:
I created a new project with the Arduino framework for the Nucleo H743 board, and it does not compile. I did not add any code to the project before trying to compile.

I have used Platformio successfully with Teensy's on this computer.

Here is the error:
```

C:\Users\user\.platformio\packages\framework-arduinoststm32\variants\STM32H7xx\H742Z(G-I)T_H743Z(G-I)T_H747A(G-I)I_H747I(G-I)T_H750ZBT_H753ZIT_H757AII_H757IIT\variant_NUCLEO_H743ZI.cpp:211:3: error: 'NC_' was not declared in this scope; did you mean 'NC'?
  211 |   NC_,
      |   ^~~
      |   NC

The terminal process "C:\Users\user\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'nucleo_h743zi'" terminated with exit code: 1.
```

kyleb-imp, 2021-07-08:
I tried compiling a fresh project for the nucleo g431kb, which was successful. So my ststm32 arduino installation seems to be good, and this seems to be an issue specific to the h743 board.

fpistm, 2021-07-09:
Hi @kyleb-imp 
yes it is a typo issue and fixed in the STM32 core:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1383

Simply remove the "_" and this will be fine.

maxgerhardt, 2021-07-09:
For better reproducibility among other computers you can **alternatively** add 

```ini
platform_packages = 
   framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/refs/heads/master.zip
```
([docs](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages)) to the project's `platformio.ini` to get the master-branch version of the Arduino core, where this issue is fixed. To only fix the compilation issue you have locally, I'd advise fpistm's method.

kyleb-imp, 2021-07-12:
@maxgerhardt Thanks that worked! So the platformio repo is manually updated with the arduino repo from time to time? Just wondering which I should be using for projects that need some amount of build process reliability.

maxgerhardt, 2021-07-12:
In the [release history](https://github.com/platformio/platform-ststm32/releases) of the platform you can see when an update to a certain Arduino core version is done. The platform (`platform-ststm32`) is updated periodically when a new **stable** release of a framework is made (or some other bug is fixed or feature is added). The [documentation](https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions) tells you how to point to a specific platform version to have a reproducable build.

Note that also with `platform_packages` you can point to a specific commit if you have `git` available in the system, e.g. with 
``` ini
platform_packages = 
   framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32.git#commithash_or_branch
```

More technical details can be found [here](https://community.platformio.org/t/arduino-due-or-other-sam3-4-upload-fails-with-atmel-ice-jtag-on-vscode-with-platformio/22258/8?u=maxgerhardt).

kyleb-imp, 2021-07-12:
Thanks I appreciate your help!

closed 2021-07-12 by kyleb-imp