platformio/platform-ststm32 #543: Board support request: NUCLEO-WL55JCx

gbThreepwood, 2021-07-09:
The board comes in two versions NUCLEO-WL55JC1, and NUCLEO-WL55JC2, where the differences is what frequency they support for the radio: https://www.st.com/en/evaluation-tools/nucleo-wl55jc.html

The board is already supported by Mbed (https://os.mbed.com/platforms/ST-Nucleo-WL55JC/), and Zephyr (https://docs.zephyrproject.org/latest/boards/arm/nucleo_wl55jc/doc/nucleo_wl55jc.html).

I have access to both versions of the board. I have managed to get them working in PlatformIO with the Mbed framework, by adding a custom "board.json" file to my boards directory.

Status:
* I have only tested STlink
* I have only tested Mbed
* I am not sure how to obtain the .svd file
* The two versions of the board requires different configuration for the LoRaWan frequency. I am not sure if this should be accounted for (e.g. by having two different board configurations), or if it should be the responsibility of the user to figure this out.

```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32WL55JC1",
    "f_cpu": "48000000L",
    "mcu": "stm32wl55jc",
    "product_line": "STM32WL55xx",
    "variant": "STM32WL55JCx/WL55JC(1-2)"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32WL55JC",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_target": "stm32wlx",
    "svd_path": "STM32F411xx.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "mbed",
    "stm32cube",
    "libopencm3",
    "zephyr"
  ],
  "name": "ST Nucleo WL55JC",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 262144,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://www.st.com/en/evaluation-tools/nucleo-wl55jc.html",
  "vendor": "ST"
}
```

I'd be happy to create a pull request as soon as I have figured out how to resolve the remaining issues.

Is there anything else that should be updated while adding a new board? Do I create a separate pull request for updating the documentation?

gbThreepwood, 2021-07-09:
I found two files "STM32WL5x_CM4.svd", and "STM32WL5x_CM0P.svd" in the installation directory for STM32CubeIDE. The files describes the registers of the M4 CPU, and the M0 CPU respectively.

The files are also available here: https://github.com/fullyautomated/st-svd

The files are under the Apache-2.0 license, thus I suppose it should be no problem to add them to the PlatformIO distribution (/platforms/ststm32/misc/svd)

gbThreepwood, 2021-07-09:
After including the "STM32WL5x_CM4.svd" file, direct manipulation of the peripheral registers appears to be working fine.

gudnimg, 2021-07-09:
You can find the SVD files here: https://www.keil.com/dd2/pack/ I recently updated the all the files  (https://github.com/platformio/platform-ststm32/pull/514) but I seem to have forgotten to add the WL series.

gbThreepwood, 2021-07-09:
Thanks. I managed to find the files after going through the hassle of installing STM32Cube, but I will bookmark your link for future reference.

I have created a pull request (#544) for the json config, and related SVD file. It works on my machine, but as this is my first contribution to PlatformIO, it could be that I have overlooked something.

gbThreepwood, 2021-07-10:
Creating new projects are not working anymore. The projects created yesterday are still functional, but I can not create new ones.

For some reason the STM32WL55 board appears under the platform "atmel avr" in addition to "ST STM32" in the list of boards in the project wizard. Adding the line `"platforms": ["ststm32"],` to the nucleo_wl55jc.json file solves that issue, but creation of new project is still broken.

gbThreepwood, 2021-07-10:
> Creating new Mbed projects are not working anymore. The projects created yesterday are still functional, but I can not create new ones.

Looks like the problem is not related to the new board, please ignore. It was only occurring when I choose to save the project outside the default location using a relatively long path for the project root. Strangely though it is not occurring anymore.

PR #544 (2021-07-27): Add support for the NUCLEO-WL55JCx
Added support for the NUCLEO-WL55JCx boards. Resolved #543
commit 9c06593e (2021-07-27): Add support for the NUCLEO-WL55JCx (#544)

Resolved #543
closed 2021-07-27 by valeros