platformio/platform-ststm32 #546: STM32duino v2.0: I get at the build with Maple Mini an Bootloader 2.0 a fatal error

Ralf9, 2021-07-17:
Hi,
with the core v2.0 and the Bootloader 2.0 I get at build a fatal error. I use Windows 10 and Visual Studio Code
```
Processing maple_mini_b20 (platform: https://github.com/platformio/platform-ststm32.git; board: maple_mini_b20; framework: arduino)
--------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/maple_mini_b20.html
PLATFORM: ST STM32 (14.0.1+sha.f702b3f) > Maple Mini Bootloader 2.0        
HARDWARE: STM32F103CBT6 72MHz, 20KB RAM, 120KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.20000.210603 (2.0.0)
 - framework-cmsis 2.50700.210515 (5.7.0)
 - tool-dfuutil 1.9.200310
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Warning! Cannot find linker script for the current target!

LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o
<command-line>: fatal error: variant_MAPLE_MINI_B20.h: No such file or directory

```

here is the platformio.ini
```
[env:maple_mini_b20]
platform = https://github.com/platformio/platform-ststm32.git
board = maple_mini_b20
board_build.core = stm32
framework = arduino
```
and here the main.cpp
```
#include <Arduino.h>

#ifndef LED_BUILTIN
  #define LED_BUILTIN 33
#endif

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

Ralf9, 2021-07-17:
when I make a build with verbose this is the command before the fatal error:
```
arm-none-eabi-gcc -o .pio\build\maple_mini_b20\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o -c -std=gnu11 -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=50101 -DSTM32F103xB -DSTM32F1xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_MAPLE_MINI_B20 -DBOARD_NAME=\"MAPLE_MINI_B20\" -DHAL_UART_MODULE_ENABLED -DUSE_FULL_LL_DRIVER -DVARIANT_H=\"variant_MAPLE_MINI_B20.h\" -DBL_LEGACY_LEAF -DVECT_TAB_OFFSET=0x2000 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\avr 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\LL 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\OpenAMP 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\hid 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb\cdc 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Inc 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F1xx_HAL_Driver\Src 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\STM32F1xx 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Inc 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\ST\STM32_USB_Device_Library\Core\Src 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\open-amp\lib\include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\libmetal\lib\include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Middlewares\OpenAMP\virtual_driver 
-IC:\Users\Ralf\.platformio\packages\framework-cmsis\CMSIS\Core\Include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Include 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F1xx\Source\Templates\gcc 
-IC:\Users\Ralf\.platformio\packages\framework-cmsis\CMSIS\DSP\Include 
-IC:\Users\Ralf\.platformio\packages\framework-cmsis\CMSIS\DSP\PrivateInclude 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\cores\arduino 
-IC:\Users\Ralf\.platformio\packages\framework-arduinoststm32\variants\MAPLEMINI_F103CB C:\Users\Ralf\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\HAL\stm32yyxx_hal_can.c
```
The problem is, in the core 2.0. the variants are in:
`variants\STM32F1xx\F103C8T_F103CB(T-U)`
and there is no "variant_MAPLE_MINI_B20.h"

I also noticed that at release 13 and 14 the arduino- and platformio version is the same. Is this ok?
`ARDUINO=10808 PLATFORMIO=50101`

In the arduino IDE with core 2.0 is
`ARDUINO=10815 `

Ralf9, 2021-08-28:
am I the only one who uses the Maple Mini with Bootloader 2.0 under platformio?

Ralf9, 2021-11-03:
With v14.0, v14.2 and v15 I have  still this error
`<command-line>: fatal error: variant_MAPLE_MINI_B20.h: No such file or directory`

platformio.ini
```
[env:maple_mini_b20_test]
platform = ststm32@14.2.0
board = maple_mini_b20
board_build.core = stm32
board_build.variant = MAPLEMINI_F103CB

framework = arduino
build_flags=
  -D USBCON
  -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
```

Ralf9, 2021-11-06:
In the meantime i was able to isolate the bug.
At the board "maple_mini_b20" and the core "stm32" the variant is not found.

In the folder "board" are in the file "maple_mini_b20.json" the variant entries incomplete.
When I add the entries **"arduino": {..** , **"framework_extra_flags": {**  and edit **"variant":** the build is success.
```
   "build": {
     "arduino": {
       "variant_h": "variant_MAPLEMINI_F103CB.h"
     },
     "core": "maple",
     "cpu": "cortex-m3",
     "extra_flags": "-DSTM32F103xB",
     "f_cpu": "72000000L",
     "framework_extra_flags": {
       "arduino": "-DARDUINO_MAPLEMINI_F103CB"
     },
     "hwids": [
       [
         "0x1EAF",
         "0x0003"
       ],
       [
         "0x1EAF",
         "0x0004"
       ]
     ],
     "mcu": "stm32f103cbt6",
     "product_line": "STM32F103xB",
     "variant": "STM32F1xx/F103C8T_F103CB(T-U)"
   },
```

here
https://github.com/platformio/platform-ststm32/blob/develop/tests/arduino-blink-different-cores/platformio.ini
is still missing a test for board "maple_mini_b20" and the core "stm32"

e.g.:
```
[env:maple_mini_b20_maple]
platform = ststm32
framework = arduino
board = maple_mini_b20

[env:maple_mini_b20_stm32]
platform = ststm32
framework = arduino
board = maple_mini_b20
board_build.core = stm32
```

valeros, 2021-11-12:
I wouldn't call this behavior a bug. This board is simply not meant to be used with the STM32Duino core, only maple. Since you specified non-default Arduino core, you should also set all other core-specific options:
```ini
[env:maple_mini_b20]
board = maple_mini_b20
board_build.core = stm32
board_build.variant = STM32F1xx/F103C8T_F103CB(T-U)
board_build.arduino.variant_h=variant_MAPLEMINI_F103CB.h
board_build.framework_extra_flags.arduino=-DARDUINO_MAPLEMINI_F103CB
```

closed 2021-11-12 by valeros