platformio/platform-ststm32 #548: Support for  STM32F427VIT6

natyshemer, 2021-07-27:
Hi, 
I was wondering if there is any plan to support the STM32F427VIT6 board?
Thank you
Naty

valeros, 2021-07-27:
Hi @natyshemer ! You can use the `armstrap_eagle2048` board, it has the same MCU.

closed 2021-07-27 by valeros

natyshemer, 2021-07-27:
Hi , this is great. 
Do you know if there any plans to support the Arduino framework with this board?

valeros, 2021-07-27:
I'd recommend you to ask in the Arduino repository.

natyshemer, 2021-07-27:
Sorry but from what I understood the Arduino framework has been archived and is now only refers to Atmel boards. Did you mean this repo?
[image]

valeros, 2021-07-27:
I meant this repo https://github.com/stm32duino/Arduino_Core_STM32