platformio/platform-ststm32 #549: After defining the string, it cannot be output through the serial port

wang5561207, 2021-08-08:
#include <SoftwareSerial.h>
SoftwareSerial mySerial1(PA10,PA9); //(rx,tx)
String d = “123123123”;
void setup() {
 mySerial1.begin(9600);
 mySerial1.print(d);
}
void loop(){
}
 
**The serial port cannot output, but char [] can**

platformio.ini

[env:genericSTM32F103C6]
platform = ststm32
board = genericSTM32F103C6
framework = arduino
upload_protocol = stlink
lib_deps = 
	adafruit/DHT sensor library@^1.4.2
	stm32duino/STM32duino FreeRTOS@^10.2.1

valeros, 2021-08-09:
Hi @wang5561207 ! Does it work in Arduino IDE?

wang5561207, 2021-08-09:
> Hi @wang5561207 ! Does it work in Arduino IDE?

Sorry, I only use vscode because my development environment does not allow me to use Arduino ide

valeros, 2021-08-10:
Unfortunately, I don't have any F103C6-based hardware at hand, but I tried your example with Nucleo-F401RE and everything works as expected. Does it work if you use the default `Serial` instance on the PA2, PA3 pins?

fpistm, 2021-08-10:
Tested with Arduino IDE 1.8.15 with a BluePill F103C8 and it is OK.

[image]

Check your wiring.

wang5561207, 2021-08-13:
> Tested with Arduino IDE 1.8.15 with a BluePill F103C8 and it is OK.
> 
> [image]
> 
> Check your wiring.

Have you tested in platform IO??

fpistm, 2021-08-13:
> Have you tested in platform IO??

No, I do not use PIO.