platformio/platform-ststm32 #552: Add support for STM32WL55 boards on arduino framework

dolfandringa, 2021-08-15:
The latest version (2.0.0) from the Arduino Core STM32 platform adds support for the STM32WL55 series of microcontrollers. Extending the support to platformio's arduino framework great. I already managed to modify the board definitions locally in git forks and get the code to compile after fixing this issue: https://github.com/stm32duino/Arduino_Core_STM32/issues/1473 as well, still need to submit the corresponding PR). I can [make a PR](https://github.com/platformio/platform-ststm32/compare/develop...dolfandringa:feature/stm32duino_stm32wl55_support?expand=1) once the Arduino Core STM32 issue has been resolved.

dolfandringa, 2021-08-15:
https://github.com/platformio/platform-ststm32/compare/develop...dolfandringa:feature/stm32duino_stm32wl55_support?expand=1 is my WIP for this. Any feedback would be appreciated

fpistm, 2021-08-15:
@dolfandringa 
STM32WL is not part of the 2.0.0, it is merged in the master and will be available with 2.1.0.
See https://github.com/stm32duino/Arduino_Core_STM32/pull/1429

And no issue need to be resolved for stm32duino/Arduino_Core_STM32#1473 

Note that currently, the LoRa is not supported for WL only basic MCU features.

dolfandringa, 2021-08-15:
OK, updated the ticket to point to 2.1.0

valeros, 2021-08-16:
Hi @dolfandringa ! Once @fpistm publish a new release this platform will updated accordingly.

dolfandringa, 2021-08-16:
@valeros shall I submit the PR for this? I managed to get a basic blink sketch compiled, uploaded and debugged on my STM32WL55 using both fork branches.

valeros, 2021-08-16:
Feel free to submit a PR, but keep in mind it won't be merged until Arduino core v2.1.0 is published.

commit b35666c6 (2021-10-31): Update Arduino core to latest v2.1.0

Resolves #552, resolves #553, resolves #567, resolves #575