platformio/platform-ststm32 #554: Update STM32CUBEF4 framework

Coolxer, 2021-08-18:
Hello,
Is there any possible way to use the newest version of STM32CUBEF4 ? 
I just made plaform update and ST STM32 platform version is fine 14.1.0.
Unfortunately requirement version of framework-stm32cubef4 is 1.25.0, but I have 1.25.2 already installed.

[image]

On the repository https://github.com/STMicroelectronics/STM32CubeF4
there is a version 1.26.2. I dont know if can i just update it manually (by cloning and replacing existing files ?)  or there any other dependencies, that also must be changed, like, this:
[image]

I really wanna update, not sure to which version will be enough, but there was introduced new function which I wanna use.

Thanks for the help.

Coolxer, 2021-08-18:
It seems to be working, after I manually changed files. I also changed the require version and installed version package.json info. I can build the project with new dependencies, which was my goal, but after i clean the project (routine activity) it stopping working as expected. There is also misbehaviour in platform infomations, list of packages doesn't show which version is actually installed. 

For summary, I found a temporary solution but looking forward for legit.

noahwilliamsson, 2021-08-19:
+1 for upgrading to 1.26.x.

I've run into an issue where the framework's `Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c` overrides my local `src/system_stm32f4xx.c`.  Before version 1.26.x, the interrupt vector address is hardcoded in the framework, which prevents me from modifying it in my local copy.

Starting with version 1.26, the default vector tab address in the framework is hidden behind an `#ifdef`, which resolve the issue I'm facing:

- https://github.com/STMicroelectronics/STM32CubeF4/blame/master/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/system_stm32f4xx.c#L90-L113

From the [release notes for 1.26.0](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Release_Notes.html):

> Protect Vector table modification following SRAM or FLASH preprocessor directive by a generic preprocessor directive : USER_VECT_TAB_ADDRESS

valeros, 2021-08-24:
Hi @Coolxer ! All STM32Cube package have been updated, please retest with the upstream version of the platform (a [Git client](https://git-scm.com/downloads) is required):
```ini
[env:your_board]
platform = https://github.com/platformio/platform-ststm32.git
framework = stm32cube
board = ...
```

Coolxer, 2021-08-24:
Thanks a lot

closed 2021-08-24 by Coolxer