platformio/platform-ststm32 #555: Update STM32CubeG4 framework to v1.4.0

t04glovern, 2021-08-20:
Hi,

Would it be possible to update the STM32CubeG4 framework published from 1.3.0 to [1.4.0](https://github.com/STMicroelectronics/STM32CubeG4/releases/tag/v1.4.0)?

[image]

We're getting the following issues with timing after making changes to our board definitions

```bash
/home/nathan/.platformio/packages/framework-stm32cubeg4/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_fmc.c:593:78: error: unknown type name 'FMC_NAND_PCC_TimingTypeDef'; did you mean 'FDCAN_Config_TypeDef'?
 HAL_StatusTypeDef FMC_NAND_CommonSpace_Timing_Init(FMC_NAND_TypeDef *Device, FMC_NAND_PCC_TimingTypeDef *Timing, uint32_t Bank)
                                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                              FDCAN_Config_TypeDef
Compiling .pio/build/nucleo_g474re/FrameworkCMSISDevice/gcc/startup_stm32g474xx.o
Compiling .pio/build/nucleo_g474re/FrameworkCMSISDevice/system_stm32g4xx.o
/home/nathan/.platformio/packages/framework-stm32cubeg4/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_fmc.c:620:55: error: unknown type name 'FMC_NAND_TypeDef'; did you mean 'FMC_Bank3_TypeDef'?
 HAL_StatusTypeDef FMC_NAND_AttributeSpace_Timing_Init(FMC_NAND_TypeDef *Device, FMC_NAND_PCC_TimingTypeDef *Timing, uint32_t Bank)
                                                       ^~~~~~~~~~~~~~~~
                                                       FMC_Bank3_TypeDef
/home/nathan/.platformio/packages/framework-stm32cubeg4/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_fmc.c:620:81: error: unknown type name 'FMC_NAND_PCC_TimingTypeDef'; did you mean 'FDCAN_Config_TypeDef'?
 HAL_StatusTypeDef FMC_NAND_AttributeSpace_Timing_Init(FMC_NAND_TypeDef *Device, FMC_NAND_PCC_TimingTypeDef *Timing, uint32_t Bank)
                                                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
                                                                                 FDCAN_Config_TypeDef
```

We can resolve these issues by manually copying the [v1.4.0 code](https://github.com/STMicroelectronics/STM32CubeG4/releases/tag/v1.4.0) over into the `.platformio/packages/framework-stm32cubeg4/Drivers` directory

Also is there an offical way to get these third-party dependencies published? I noticed that you can define a version in [platform.json](https://github.com/platformio/platform-ststm32/blob/develop/platform.json#L267) however it seems like the package needs to be published by an someone internal to the platformIO repository before use.

```json
    "framework-stm32cubeg4": {
      "type": "framework",
      "optional": true,
      "owner": "platformio",
      "version": "~1.3.0"
    },
```

Thank you for any help you can provide!

Regards,

valeros, 2021-08-24:
Hi @t04glovern ! All STM32Cube package have been updated to the latest available, please retest with the upstream version of the platform (a [Git client](https://git-scm.com/downloads) is required):
```ini
[env:your_board]
platform = https://github.com/platformio/platform-ststm32.git
framework = stm32cube
board = ...
```

t04glovern, 2021-08-24:
Excellent! Thanks so much, works great one my end.

closed 2021-08-24 by t04glovern