platformio/platform-ststm32 #556: Update ST32CubeG0 framework to v1.5.0

djix123, 2021-08-22:
Framework is currently 1.3.0 and does not support latest G0 series processors: e.g. STM32G0B1 / STM32G0C1.

valeros, 2021-08-24:
Hi @djix123 ! All STM32Cube package have been updated, please retest with the upstream version of the platform (a [Git client](https://git-scm.com/downloads) is required):
```ini
[env:your_board]
platform = https://github.com/platformio/platform-ststm32.git
framework = stm32cube
board = ...
```

djix123, 2021-08-24:
Thanks - that appeared to work.

I needed to delete the prior version of 'https://github.com/platformio/platform-ststm32.git' I already had in the platforms directory, and I also deleted the old stm32cubeG0 framework in the packages directory.

After that - all good. Without doing the deletions it didn't appear to download the latest versions.