platformio/platform-ststm32 #557: Arduino Portenta H7 Wifi

alvarolb, 2021-08-23:
I cannot make a simple sketch with WiFi to work on Arduino Portenta. 

`platform.ini`
```ìni
[env:portenta_h7_m4]
platform = ststm32
board = portenta_h7_m4
framework = arduino
```

`main.cpp`
```cpp
#include <WiFi.h>

///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = "";        // your network SSID (name)
char pass[] = "";        // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                 // your network key Index number (needed only for WEP)

int status = WL_IDLE_STATUS;
// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
// IPAddress server(93,184,216,34);  // IP address for example.com (no DNS)
char server[] = "example.com";       // host name for example.com (using DNS)

WiFiClient client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  // attempt to connect to Wifi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 3 seconds for connection:
    delay(3000);
  }
  Serial.println("Connected to wifi");
  printWifiStatus();

  Serial.println("\nStarting connection to server...");
  // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.println("connected to server");
    // Make a HTTP request:
    client.println("GET /index.html HTTP/1.1");
    client.print("Host: ");
    client.println(server);
    client.println("Connection: close");
    client.println();
  }
}

void loop() {
  // if there are incoming bytes available
  // from the server, read them and print them:
  while (client.available()) {
    char c = client.read();
    Serial.write(c);
  }

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting from server.");
    client.stop();

    // do nothing forevermore:
    while (true);
  }
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

Error

```bash
src/main.cpp: In function 'void setup()':
src/main.cpp:58:3: error: 'printWifiStatus' was not declared in this scope
   printWifiStatus();
   ^~~~~~~~~~~~~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp: In static member function 'static char SpiDrv::spiTransfer(char)':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: error: 'F_CPU' was not declared in this scope
```

I think that the problem is that it takes a bad `WiFi` library, the default one in the Arduino Core. 

```bash
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.2.7
|   |-- <SPI>
```

Moreover, in other sketch it selects `WiFiNINA` as the library for wifi... which does not apply. This board should use the libraries from the [ArduinoCore-mbed](https://github.com/arduino/ArduinoCore-mbed/tree/master/libraries) by default. 

Is there any way to make it working? I tried with lib_compact_mode, extra lib dirs... but I cannot make it work.

Thanks.

valeros, 2021-08-24:
Hi @alvarolb ! You need to either forward declare the `printWifiStatus` function at the top of the file or move the entire function definition above the `setup` function.

alvarolb, 2021-08-24:
Hi, I think this issue should not be closed. Forget `printWifiStatus`, there are other errors associated with the library. From the example:

```bash
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: error: 'F_CPU' was not declared in this scope
```

alvarolb, 2021-08-24:
Full error while moving `printWifiStatus` to correct place:

```bash
Processing portenta_h7_m4 (platform: ststm32; board: portenta_h7_m4; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m4.html
PLATFORM: ST STM32 (14.1.0) > Arduino Portenta H7 (M4 core)
HARDWARE: STM32H747XIH6 480MHz, 287.35KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed 2.3.1 
 - tool-dfuutil 1.9.200310 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.2.7
|   |-- <SPI>
Building in release mode
Compiling .pio/build/portenta_h7_m4/src/main.cpp.o
Compiling .pio/build/portenta_h7_m4/libfc7/SPI/SPI.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/WiFi.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/WiFiClient.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/WiFiServer.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/WiFiUdp.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/utility/server_drv.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/utility/spi_drv.cpp.o
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/utility/wifi_drv.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/Interrupts.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/Serial.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/USB/PluggableUSBDevice.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/USB/USBCDC.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/USB/USBSerial.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/abi.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/api/Common.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/api/IPAddress.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/api/PluggableUSB.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/api/Print.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/api/Stream.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/api/String.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/arm_hal_random.c.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/as_mbed_library/variant.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/itoa.c.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/main.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/mbed/platform/cxxsupport/mstd_mutex.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/pinToIndex.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/wiring.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/wiring_analog.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/wiring_digital.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pio/build/portenta_h7_m4/FrameworkArduino/wiring_shift.cpp.o
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp: In static member function 'static char SpiDrv::spiTransfer(char)':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: error: 'F_CPU' was not declared in this scope
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: in definition of macro 'DELAY_SPI'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^~~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:76:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: suggested alternative: 'FPU'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^
[... 36 lines cut ...]
         {
         ^
In file included from .pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:22:0:
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.h: At global scope:
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.h:41:13: warning: 'initialized' defined but not used [-Wunused-variable]
 static bool initialized = false;
             ^~~~~~~~~~~
*** [.pio/build/portenta_h7_m4/libb6a/WiFi/utility/spi_drv.cpp.o] Error 1
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/wifi_drv.cpp: In static member function 'static int8_t WiFiDrv::startScanNetworks()':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/wifi_drv.cpp:417:19: warning: comparison is always false due to limited range of data type [-Wtype-limits]
     return (_data == WL_FAILURE)? _data : WL_SUCCESS;
             ~~~~~~^~~~~~~~~~~~~
[... cut ...]
          ^~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/wifi_drv.cpp: In static member function 'static int32_t WiFiDrv::getRSSINetoworks(uint8_t)':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/wifi_drv.cpp:476:10: warning: converting to non-pointer type 'int32_t {aka long int}' from NULL [-Wconversion-null]
   return NULL;
          ^~~~
================================================================= [FAILED] Took 2.61 seconds =================================================================
```

valeros, 2021-08-24:
Could you please retest with the upstream version of the platform (a [Git client](https://git-scm.com/downloads) is required):
```ini
[env:portenta_h7_m4]
platform = https://github.com/platformio/platform-ststm32.git
framework = arduino
board = portenta_h7_m4
```

alvarolb, 2021-08-24:
Hi! I got the same error.

valeros, 2021-08-24:
Please attach here the build log with the upstream version.

alvarolb, 2021-08-24:
```bash
Processing portenta_h7_m4 (platform: https://github.com/platformio/platform-ststm32.git; framework: arduino; board: portenta_h7_m4)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/portenta_h7_m4.html
PLATFORM: ST STM32 (14.1.0+sha.9ea371b) > Arduino Portenta H7 (M4 core)
HARDWARE: STM32H747XIH6 480MHz, 287.35KB RAM, 1MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduino-mbed 2.4.1 
 - tool-dfuutil 1.9.200310 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.2.7
|   |-- <SPI>
Building in release mode
Archiving .pio/build/portenta_h7_m4/libfc7/libSPI.a
Indexing .pio/build/portenta_h7_m4/libfc7/libSPI.a
Compiling .pio/build/portenta_h7_m4/libb6a/WiFi/utility/spi_drv.cpp.o
Archiving .pio/build/portenta_h7_m4/libFrameworkArduinoVariant.a
Indexing .pio/build/portenta_h7_m4/libFrameworkArduinoVariant.a
Archiving .pio/build/portenta_h7_m4/libFrameworkArduino.a
Indexing .pio/build/portenta_h7_m4/libFrameworkArduino.a
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp: In static member function 'static char SpiDrv::spiTransfer(char)':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: error: 'F_CPU' was not declared in this scope
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: in definition of macro 'DELAY_SPI'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^~~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:76:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: suggested alternative: 'FPU'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: in definition of macro 'DELAY_SPI'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^~~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:76:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp: In static member function 'static void SpiDrv::getParam(uint8_t*)':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: error: 'F_CPU' was not declared in this scope
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: in definition of macro 'DELAY_SPI'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^~~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:148:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: suggested alternative: 'FPU'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:36:79: note: in definition of macro 'DELAY_SPI'
 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }
                                                                               ^~~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:148:5: note: in expansion of macro 'DELAY_TRANSFER'
     DELAY_TRANSFER();
     ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp: In static member function 'static int SpiDrv::waitResponseCmd(uint8_t, uint8_t, uint8_t*, uint8_t*)':
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:160:36: warning: suggest braces around empty body in an 'else' statement [-Wempty-body]
         CHECK_DATA(numParam, _data);
                                    ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:127:10: warning: this 'else' clause does not guard... [-Wmisleading-indentation]
         }else                                           \
          ^
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:127:10: note: in definition of macro 'CHECK_DATA'
         }else                                           \
          ^~~~
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:161:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'else'
         {
         ^
In file included from .pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.cpp:22:0:
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.h: At global scope:
.pio/libdeps/portenta_h7_m4/WiFi/src/utility/spi_drv.h:41:13: warning: 'initialized' defined but not used [-Wunused-variable]
 static bool initialized = false;
             ^~~~~~~~~~~
*** [.pio/build/portenta_h7_m4/libb6a/WiFi/utility/spi_drv.cpp.o] Error 1
============================================== [FAILED] Took 1.63 seconds ==============================================
```

valeros, 2021-08-24:
OK, so the problem here is that PlatformIO picks a wrong `WiFi` library. The `WiFiNINA` library is not the part of the framework so 
where it's installed then? Is it a part of your project?

alvarolb, 2021-08-24:
It took `WiFiNINA` in a more complex project. So I tried to get a minimal working example like posted before. With this example it always finds `WiFi` library in its `1.2.7` version, [this one](https://github.com/arduino-libraries/WiFi), the default one for the Arduino WiFi shield. The WiFi library from mbed its in [version 1.0](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/WiFi/library.properties).

In this project I have nothing special:

[image]

valeros, 2021-08-24:
In a nutshell, it's expected behavior as libraries inside your project have a higher priority over libraries inside framework. 

> It took WiFiNINA in a more complex project. 

I suppose it's still the same project, you just have two different environments that depend on different `WiFi` libraries? You didn't provide your real `platformio.ini` file so I can only guess how you install that external `WiFi` library v1.2.7, I tried to reproduce your issue using the following configuration but to no avail:
```ini
[env:portenta_h7_m4]
platform = ststm32
framework = arduino
board = portenta_h7_m4
lib_deps =
    arduino-libraries/WiFi @ ^1.2.7
    
[env:portenta_h7_m4_libs_ignored]
platform = ststm32
framework = arduino
board = portenta_h7_m4
```

alvarolb, 2021-08-24:
Thanks for pointing me to the correct direction. Now I have realized that in a first instance I had a `libs_extra_dirs` pointing to some local libraries where it is installed `WiFiNiNa` among others... I later removed that for the basic example, but it seems that something got installed/cached there. I have started from scratch and it is compiling fine. Thanks for your time @valeros.

Is it expected that libraries from  `lib_extra_dirs` replace the ones from the framework?

closed 2021-08-24 by alvarolb

valeros, 2021-08-24:
> Is it expected that libraries from lib_extra_dirs replace the ones from the framework?

Libraries from any framework have the lowest priority. You can find more info on the LDF and storage priorities in the docs https://docs.platformio.org/en/latest/librarymanager/ldf.html#storage