platformio/platform-ststm32 #564: Add support for multiple ST-Link adapters

jcw, 2021-09-29:
I've set up a Raspberry Pi as build farm for STM32 µC boards. The problem is that the ST-Link interfaces on these boards all look the same, apart from the serial number, exposed in the USB descriptors. For OpenOCD, which is the default upload tool, an extra parameter needs to be specified (`hla_serial <serialnum>`).

Attached is a `platformio.ini` file which properly uploads and connect to the matching serial port. It works, but it's messy, and I need to specify the board type again, even though PIO already has that info in its board definition file. Also included is a `main.cpp`, to test the builds / uploads (which will obviously fail at some point, without these exact same boards present).

[blink.zip](https://github.com/platformio/platformio-core/files/7007132/blink.zip)

It would be useful to streamline this all a bit more, perhaps with a new `upload_serial` config parameter. For the serial connection, a second simplification would be to use this same serial number to match the proper port (in `minicom`?) when `monitor_port` is not set. That way, all you'd need to pick a _specific_ ST-Link would be to add `upload serial = ...`.

(see also [my forum post](https://community.platformio.org/t/choosing-stlink-v2-programmer/10716/4) on this topic)

ivankravets, 2021-09-29:
Yes, I think we can parse `upload_protocol`. Need to discuss a possible format https://openocd.org/doc/html/Debug-Adapter-Configuration.html

jcw, 2022-04-26:
Been a while, I'm going to investigate the `extra_script.py` approach. But there is still one issue with this - the names for ST-Link v2.1 and v3 differ:

```
monitor_port = /dev/serial/by-id/usb-STMicroelectronics_STM32_STLink_0670FF504955857567192657-if02
monitor_port = /dev/serial/by-id/usb-STMicroelectronics_STLINK-V3_004200283137510739383538-if02
```

One option would be to look at the devices which are currently present, and look for a match on their path names, perhaps something like this - untested:
```
p = os.listdir(f'/dev/serial/by-id/usb-STM*_${serno}-if02')[0]
```

jcw, 2022-04-27:
The following works (note the `findex+2`):

```
Import("env")
try:
    findex = env.get("UPLOADERFLAGS").index("-f")
    sernum = env.GetProjectOption("custom_sernum")
    env["UPLOADERFLAGS"].insert(findex+2, ["-c", f"hla_serial {sernum}"])
except:
    pass
```

I'm trying to also auto-generate the proper monitor port setting for Linux, given that all the info is present when `custom_sernum` has been specified, but I can't quite figure out how/where to set this. I imported `projenv` and looked at its `Dump()`, but there's no mention of a `monitor_port`.

What I'm after, is to make everything upload _and_ connect with just these settings for each attached ST-Link board:

```
[env:l412]
board = nucleo_l412kb
build_flags = ${env.build_flags} -DSTM32L4
custom_sernum = 0668FF575051717867043125
```

I can figure out how to scan the `/dev/serial/by_id/` area in `extra_script.py` to find the exact path, but is there a way to generate the monitor port choice in the `post:extra_script.py` ?

Another approach would be for PIO to allow specifying a wildcard in its `monitor_port` setting:

```
custom_sernum = 0668FF575051717867043125
monitor_port = /dev/serial/by_id/usb-STM*${this.custom_sernum}-if02
```

Although that's slightly less convenient, because you then still have to specify that extra line for each board.

ivankravets, 2023-06-08:
There is a related feature request https://github.com/platformio/platformio-core/issues/4460

Please vote ( 👍  ) for it.