platformio/platform-ststm32 #576: STM32L552ZEQ don't work with last version 15.0.0.

matte0080, 2021-11-01:
I started stm32 project with version 14.2.0. It was ok. After update platform to 15.0.0 my project not compiled i receive this error:

`embedded:startup.tcl:26: Error: Can't find board/st_nucleo_l5.cfg.cfg`

This is my platform.io:
```
[env:nucleo_l552ze_q]
platform = ststm32
board = nucleo_l552ze_q
framework = arduino
lib_deps = 
	vshymanskyy/TinyGSM@^0.11.4
	arduino-libraries/ArduinoHttpClient@^0.4.0
	winlinvip/SimpleDHT@^1.0.15
	stm32duino/STM32duino Low Power@^1.1.1
	stm32duino/STM32duino RTC@^1.1.1
	bblanchon/ArduinoJson@^6.18.5
	arduino-libraries/SD@^1.2.4
	bblanchon/StreamUtils@^1.6.1
	paulstoffregen/Time@^1.6.1
	bogde/HX711@^0.7.4
	mathertel/OneButton@^2.0.3
```

Error Screenshot:
[image]

Any idea?

maxgerhardt, 2021-11-02:
Can you locally find the file `C:\Users\<user>\.platformio\platforms\ststm32\boards\nucleo_l552ze_q.json` and edit

https://github.com/platformio/platform-ststm32/blob/974571843c1128d4124941a817cee5d785eb1ef7/boards/nucleo_l552ze_q.json#L22

that is. remove the `.cfg` part in the string?

valeros, 2021-11-02:
Thanks for reporting, fixed in the dev branch. You can follow @maxgerhardt advice or install the upstream version of the platform https://docs.platformio.org/en/latest/platforms/ststm32.html#upstream

commit 321e4c7d (2021-11-02): Fix OpenOCD board name for nucleo_l552ze_q

Resolve #576
closed 2021-11-02 by valeros