platformio/platform-ststm32 #579: code is not running with Fysetc board but runs is I choose another board in platformio.ini

deladriere, 2021-11-13:
I want to blink the flash LED as a test.
```
#include <Arduino.h>

void setup()
{
    pinMode(PD3, OUTPUT);
}

void loop()
{
    digitalWrite(PD3, LOW);
    delay(1000);
    digitalWrite(PD3, HIGH);
    delay(1000);
}
```
I can upload with success using ST-LINK
but it won't run.

Mine platformio.ini is  :
```
[env:fysetc_s6]
platform = ststm32
board = fysetc_s6

framework = arduino
```
If I use this one

````
[env:mycode]

platform = ststm32
board = rumba32_f446ve
;board = fysetc_s6
framework = arduino
````
the code runs 
any idea what is wrong here ?

valeros, 2021-11-16:
Hi @deladriere ! The only obvious difference is that the manifest file for `fysetc_s6` takes into account a flash memory offset  required for an internal bootloader. Does your board have this bootloader?

deladriere, 2021-11-16:
@valeros, thanks a lot!
No, I guess I erased it because using stlink without offset. 
For the moment, I don't need their bootloader.
I thought the offset would only apply for the DFU upload ?
So if erased, my code should work? Or maybe I have to specify something on the platformio.ini?

valeros, 2021-11-16:
I believe you can get around the problem by specifying the default offset `0x08000000` for the STM32 MCUs, something like this:
```ini
[env:fysetc_s6]
platform = ststm32
framework = arduino
board = fysetc_s6
board_upload.offset_address = 0x08000000
```

deladriere, 2021-11-18:
@valeros, thanks. This worked ! ... but why do we need this offset? (sorry if this is a rookie question)

valeros, 2021-11-18:
In a nutshell, this offset is only required if you want to use the `fysetc_s6` board and accidentally erased the bootloader.  By default, the bootloader resides in a flash memory area from `0x08000000` to `0x08010000`. The [board manifest](https://github.com/platformio/platform-ststm32/blob/develop/boards/fysetc_s6.json#L41) for  `fysetc_s6` takes into account this bootloader and hence PlatformIO uploads your application precisely after the bootloader.

closed 2021-11-18 by valeros