platformio/platform-ststm32 #580: Arduino-GenericSTM32f103C8 pinMode change behaviour on others pins.

KoolElmout, 2021-11-15:
Hello,
 I cannot find any issue on this forum so i create a new one.
On PlatformIO (Core 5.2.3 - Home 3.4.0), i reduce my code to a simple blink test with two pins and run it on a bluepill powered by stlink debugger. (So 3.3V) I also test is power by USB but same behavior.
If i comment line AA: PB11 is switching between level 0V<->3.3V which is normal
If i un-comment line AA : PB11 is switching between level 0V<->2V which is abnormal. It is as if the pull up configuration is changed on PB10 for a faulty.

I tried several other pin mode change and it seem related to PB10 declared as output. On google i found out that the mapple core is faulty but i do not this core here. 

```
#include <Arduino.h> 
void setup() {
    digitalWrite(PB11, LOW);
    pinMode(PB11, OUTPUT);
    digitalWrite(SC_RST_PIN, HIGH);
    pinMode(PB10, OUTPUT); //line AA
}
void loop() {
    digitalWrite(PB11, HIGH); //Line BB
    delay(100);
    digitalWrite(PB11, LOW);
    delay(100);
}
```

Here is the platformio.ini
```[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol=stlink
debug_tool=stlink
```

valeros, 2021-11-15:
Hi @KoolElmout ! I believe `pinMode` should be called set before `digitalWrite`? Anyway, does it work correctly in the Arduino IDE?

KoolElmout, 2021-11-16:
Thank, I will have a tryout on Arduino IDE.
Usually i made a digital write before turning on the output function to have directly the good voltage at the pin mode change. It works well (arduino/stm32/microchip). (It is intends to program the output register before switching on the output multiplexer of the chip). 
Regards.

KoolElmout, 2021-11-17:
I have tryout on Arduino IDE 1.8, with the same behavior. So i borrow another bluepill from a friend and do the same test. The trouble follow the bluepill. So this is an STM32's hardware failure and not a software problem. 
Thank again for your suggestion. I will close this issue.

closed 2021-11-17 by KoolElmout