platformio/platform-ststm32 #583: Error undefined reference to `__aeabi_uldivmod’ when build in VSCode with Zephyr 2.7.0 and stm32l476g_disco

fridaprog, 2021-11-18:
I've updated my VSCode + PlatformIO development environment to the Zephyr 2.7.0 just released some days ago. After this change my program cannot be build correctly as before due to a link error:
undefined reference to `__aeabi_dcmpun'
this unresolved comes out using the sqrt() math function. Removing the call to sqrt() my program can be build correctly.

To reproduce the problem compile this program inside VSCode (main.c):

> #include <zephyr.h>
> #include <math.h>
> void main(void)
> {
> float a = sqrt(1.23f);
> }

prj.conf contains:

> CONFIG_NEWLIB_LIBC=y

platformio.ini contains:

> [env:disco_l476vg]
> platform = ststm32
> board = disco_l476vg
> framework = zephyr

Logs and console output:

> Verbose mode can be enabled via -v, --verbose option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_l476vg.html
> PLATFORM: ST STM32 (15.0.0) > ST 32L476GDISCOVERY
> HARDWARE: STM32L476VGT6 80MHz, 128KB RAM, 1MB Flash
> DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
> PACKAGES:
> 
>     framework-zephyr 2.20700.211026 (2.7.0)
>     tool-cmake 3.21.3
>     tool-dtc 1.4.7
>     tool-gperf 3.0.4
>     tool-ninja 1.7.1
>     toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
>     Reading CMake configuration...
>     Parsing system calls...
>     Generating syscall files
>     Generating KObject files...
>     Validating driver...
>     LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
>     LDF Modes: Finder ~ chain, Compatibility ~ soft
>     Found 0 compatible libraries
>     Scanning dependencies...
>     No dependencies
>     Building in release mode
>     Compiling .pio/build/disco_l476vg/offsets/zephyr/arch/arm/core/offsets/offsets.c.o
>     Compiling .pio/build/disco_l476vg/zephyr_final/zephyr/misc/empty_file.c.o
>     ...
>     Archiving .pio/build/disco_l476vg/zephyr/kernel/libkernel.a
>     Indexing .pio/build/disco_l476vg/zephyr/kernel/libkernel.a
>     Linking .pio/build/disco_l476vg/zephyr/firmware-pre.elf
>     /home/alfredo/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.bfd: /home/alfredo/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libm.a(lib_a-w_sqrt.o): in function sqrt':** w_sqrt.c:(.text.sqrt+0x26): **undefined reference to __aeabi_dcmpun'
>     collect2: error: ld returned 1 exit status
>     *** [.pio/build/disco_l476vg/zephyr/firmware-pre.elf] Error 1

Environment:

> OS: Linux
> Toolchain: gccarmnoneeabi 1.80201.181220 (8.2.1)
> Zephyr version: 2.7.0
> Development environment: VSCode 1.62.2
> PlatformIO: Core 5.2.3 Home 3.4.0
> Framework: ST STM32 CMSIS 15.0.0

Compiling the program outside VSCode, using the Zephyr SDK 0.13.0, the link error does not occur.

valeros, 2021-11-19:
Hi @fridaprog ! Could you please update the `framework-zephyr` package to the latest and try again? Just run `pio platform update ststm32` in your IDE terminal.

fridaprog, 2021-11-19:
Hi @valeros, it works! The update operation produced this log:

```
pio platform update ststm32
Platform ststm32
--------
Updating platformio/ststm32                   15.0.0                             [Up-to-date]
Updating platformio/framework-zephyr          2.20700.211026 @ ~2.20700.0        [Updating to 2.20700.211118]
Tool Manager: Installing platformio/framework-zephyr @ 2.20700.211118
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-zephyr @ 2.20700.211118 has been installed!
Tool Manager: Removing framework-zephyr @ 2.20700.211026
Tool Manager: framework-zephyr @ 2.20700.211026 has been removed!
Updating platformio/tool-stm32duino           1.0.1 @ ~1.0.1                     [Up-to-date]
Updating platformio/tool-openocd              2.1100.211028 @ ~2.1100.0          [Up-to-date]
Updating platformio/tool-dfuutil              1.9.200310 @ ~1.9.190708           [Up-to-date]
Updating platformio/tool-cmake                3.21.3 @ ~3.21.0                   [Up-to-date]
Updating platformio/tool-dtc                  1.4.7 @ ~1.4.7                     [Up-to-date]
Updating platformio/tool-ninja                1.7.1 @ ^1.7.0                     [Up-to-date]
Updating platformio/tool-gperf                3.0.4 @ ^3.0.0                     [Up-to-date]

```

It's important to specify that the pio command has to be used inside _PlatformIO Termnal_ (starting it from VSCode status bar) and not in the normal _IDE terminal_, where probably the path to $HOME/.platformio/penv/bin/pio is not set.

Thank you very much, valeros.

closed 2021-11-22 by valeros