platformio/platform-ststm32 #590: pio run -t menuconfig ends in loop

lexologe, 2021-12-16:
Hello,

I created a STM32 Nucleo based project based.

```
pio project init --board nucleo_g474re --project-option "framework=mbed"`
```

If I try to run `pio run -t menuconfig` it ends in a loop.

```
> Executing task: platformio run --target menuconfig --environment nucleo_g474re <

Processing nucleo_g474re (platform: ststm32; board: nucleo_g474re; framework: zephyr)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_g474re.html
PLATFORM: ST STM32 (15.1.0) > Nucleo G474RE
HARDWARE: STM32G474RET6 170MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-zephyr 2.20700.211118 (2.7.0) 
 - tool-cmake 3.21.3 
 - tool-dtc 1.4.7 
 - tool-gperf 3.0.4 
 - tool-ninja 1.7.1 
 - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Running menuconfig...
[0/1] Re-running CMake...
-- Application: /home/asc/Documents/PlatformIO/Projects/bootest/zephyr
-- Zephyr version: 2.7.0 (/home/asc/.platformio/packages/framework-zephyr)
-- Board: nucleo_g474re
-- Cache files will be written to: /home/asc/.cache/zephyr
-- Found dtc: /home/asc/.platformio/packages/tool-dtc/dtc (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/home/asc/.platformio/packages/toolchain-gccarmnoneeabi@1.80201.181220)
-- Found BOARD.dts: /home/asc/.platformio/packages/framework-zephyr/boards/arm/nucleo_g474re/nucleo_g474re.dts
-- Generated zephyr.dts: /home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/dts.cmake
Parsing /home/asc/.platformio/packages/framework-zephyr/Kconfig
Loaded configuration '/home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/.config'
No change to configuration in '/home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/.config'
No change to Kconfig header in '/home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/asc/Documents/PlatformIO/Projects/bootest/.pio/build/nucleo_g474re
[0/1] Re-running CMake...
```
(the log above is copied from the terminal in VSCode. A quick test in VSCode ends in the same loop.)

--
Alex

lexologe, 2021-12-16:
The project compiles fine with `pio run`

jomiq, 2021-12-18:
I'd like to report this issue as well. Same configuration, hardware slightly different:
```
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f746zg.html
PLATFORM: ST STM32 (15.1.0) > ST Nucleo F746ZG
HARDWARE: STM32F746ZGT6 216MHz, 320KB RAM, 1MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, cmsis-dap, jlink)
PACKAGES: 
 - framework-zephyr 2.20700.211118 (2.7.0) 
 - tool-cmake 3.21.3 
 - tool-dtc 1.4.7 
 - tool-gperf 3.0.4 
 - tool-ninja 1.7.1 
 - toolchain-gccarmnoneeabi 1.80201.181220 (8.2.1)
 ```