platformio/platform-ststm32 #591: Make FPU and CMSIS-DSP configuration easier

maxgerhardt, 2022-01-04:
When using a minimal `platformio.ini` of

```ini
[env:nucleo_l476rg]
platform = ststm32
board = nucleo_l476rg
framework = stm32cube
```

with `src/main.c` of

```c
#include "stm32l4xx_hal.h"
#include <arm_math.h>

int main(void)
{
  HAL_Init();

  float32_t cosOutput;
  float32_t cosSquareOutput;
  while(1) {
     cosOutput = arm_cos_f32(1.2345f);
     arm_mult_f32(&cosOutput, &cosOutput, &cosSquareOutput, 1);
  }
}

void SysTick_Handler(void)
{
  HAL_IncTick();
}
```

Linking will fail.

```
Linking .pio\build\nucleo_l476rg\firmware.elf
.pio/build/nucleo_l476rg/src/main.o: In function `main':
main.c:(.text.startup.main+0xa): undefined reference to `arm_cos_f32'
main.c:(.text.startup.main+0x18): undefined reference to `arm_mult_f32'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nucleo_l476rg\firmware.elf] Error 1
============[FAILED] Took 5.45 seconds ===============
```

This is because the STM32Cube builder script does not add the `-larm_cortex...` switch to actually link in the library.

Adding in 
```ini
build_flags =
  -larm_cortexM4l_math
```

fixes that,

```
RAM:   [          ]   0.0% (used 44 bytes from 98304 bytes)
Flash: [          ]   0.4% (used 4660 bytes from 1048576 bytes)
Building .pio\build\nucleo_l476rg\firmware.bin
===============[SUCCESS] Took 5.48 seconds =============
```

but it's unintuitive and [undocumented](https://docs.platformio.org/en/latest/frameworks/stm32cube.html#custom-dsp-library). 

Further, above I explicitly linked in `arm_cortexM4l_math` (softfloat) instead of `arm_cortexM4lf_math` (hardfloat). This is because the STM32Cube builder script does not add any `-mfpu` or `-mfloat-abi` flags by default ([source](https://github.com/platformio/platform-ststm32/blob/2f7e82f6a5c52372c8d211504ac6ba9ffece515f/builder/frameworks/stm32cube.py#L219-L284) in the compilation process, so everything is compiled as default softfloat, which leads to a ton of linker errors if a user attempts to link with the hardfloat library

```
Linking .pio\build\nucleo_l476rg\firmware.elf
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: error: C:/Users/Max/.platformio/packages/framework-stm32cubel4/Drivers/CMSIS/DSP/Lib/GCC\libarm_cortexM4lf_math.a(arm_cos_f32.o) uses VFP register arguments, .pio/build/nucleo_l476rg/firmware.elf does not
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:/Users/Max/.platformio/packages/framework-stm32cubel4/Drivers/CMSIS/DSP/Lib/GCC\libarm_cortexM4lf_math.a(arm_cos_f32.o)
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: error: C:/Users/Max/.platformio/packages/framework-stm32cubel4/Drivers/CMSIS/DSP/Lib/GCC\libarm_cortexM4lf_math.a(arm_mult_f32.o) uses VFP register arguments, .pio/build/nucleo_l476rg/firmware.elf does not
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:/Users/Max/.platformio/packages/framework-stm32cubel4/Drivers/CMSIS/DSP/Lib/GCC\libarm_cortexM4lf_math.a(arm_mult_f32.o)
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: error: C:/Users/Max/.platformio/packages/framework-stm32cubel4/Drivers/CMSIS/DSP/Lib/GCC\libarm_cortexM4lf_math.a(arm_common_tables.o) uses VFP register arguments, .pio/build/nucleo_l476rg/firmware.elf does not
c:/users/max/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:/Users/Max/.platformio/packages/framework-stm32cubel4/Drivers/CMSIS/DSP/Lib/GCC\libarm_cortexM4lf_math.a(arm_common_tables.o)
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\nucleo_l476rg\firmware.elf] Error 1
=================[FAILED] Took 5.38 seconds ================
```

One has to explicitly add an extra script to add in `-mfloat-abi=hard -mfpu=fpv4-sp-d16` in the `CCFLAGS` and `LINKFLAGS` to make this work (see https://community.platformio.org/t/error-object-uses-vfp-register-arguments-firmware-elf-does-not/25263). 

The STM32Cube builder script should provide easier coniguration of the selected CMSIS-DSP library to be linked in, as well as being able to add hardfloat flags for those targets that support it (Cortex-M4, Cortex-M7, Cortex-M33 etc). These options should further be documented.

jnz86, 2022-08-23:
I have to agree with Max (again!). 

I am encountering this in the wild right now. Using STM32 but not the framework. 

>but it's unintuitive 

Very. I'm looking into why the linker wouldn't work as expected, but if Max hadn't made the post he did, and the user who first had the issue hadn't been clear enough for me to find it in search results I would be near-defeated. I'm a veteran programmer, but new PIO user. Making a script to add flags would be so far down the list of things I tried on my own that I would have gone back to STM32CubeIDE.

chintal, 2025-01-18:
The current state is not intuitive and difficult to search for solutions and to reason about. 

I have also used the script to get hardfloat working, though I would prefer one of the following options instead: 

1. `build_flags` in `platformio.ini` should apply to the link stage as well. Specifically, setting `-mfpu=fpv5-d16 -mfloat-abi=hard` in the `build_flags` should apply to the linker as well.  
2. Add a `link_flags` option to `platformio.ini`, and highlight in the documentation that `build_flags` does not apply to the link stage.

stevenbell, 2026-03-06:
+1 for this.  I'm new to the internals of PlatformIO, but I'm a little baffled why there isn't a combination of flags in `platformio.ini` to make this work.  It seems like an explicit `link_flags` option would simplify a lot of things.

A related use-case is that I'd like to remove the `--specs=nano.specs` line in the default CMSIS builder with something like `linker_unflags=--specs=nano.specs`.