platformio/platform-ststm32 #594: Update libopencm3 framework

qosch, 2022-02-04:
I am having issues building a simple program using the libopencm3 framework and someone on the forum concluded this is due to an old version of libopencm3 (Thread: https://community.platformio.org/t/compiler-doesnt-find-function-definition-libopencm3/25795).
Could this be updated?
Thanks :)

larmicfre, 2022-03-28:
Since custom packages are possible in PlatformIO, I've forked the libopencm3 repository and added the required package.json file. The repo is up to date and seem to be working. I'll fetch upstream upon new commits.

To use my forked libopencm3 in your build environment, you'll need to specify `platform_packages` in your `platformio.ini`. Copy and paste the following into your `platformio.ini`.

 `platform_packages = framework-libopencm3 @ https://github.com/larmicfre/libopencm3.git`

Don't forget to clean your build environment.

larmicfre, 2022-03-28:
It's atually possible to use custom packages in your enviroment, defined in your `platformio.ini`. I've forked libopencm3 library and added the required packages.json file. I'll fetch upstream upon new commits.

To use the latest libopencm3 library in your project, open `platformio.ini` and paste the following line.
`platform_packages = framework-libopencm3 @ https://github.com/larmicfre/libopencm3.git`

Don't forget to clean your enviroment.