platformio/platform-ststm32 #595: Upload with dfu fails on macOS 12 (libusb not found)

manuelbl, 2022-02-04:
Uploading the firmware using the setting `upload_protocol = dfu` fails. The error message is:

```
...
Building .pio/build/blackpill-f401cc/firmware.bin
Adding dfu suffix to firmware.bin
dyld[74082]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: /Users/someuser/.platformio/packages/tool-dfuutil/bin/dfu-suffix
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file)
*** [.pio/build/blackpill-f401cc/firmware.bin] Error -6
```
This is on macOS 12.1 with an Apple M1 CPU and on MacOS 12.2 with an Intel CPU. It does not affect macOS 11.6 (with an Intel CPU).

The package *tool-dfuutil* contains a matching version `libusb-1.0.0.dylib` in the *lib* directory, but it's not found as the executable contains a fixed path to `/usr/local/opt/libusb/lib/libusb-1.0.0.dylib`. In macOS 11, this was probably treated differently.

Installing *libusb* separately will not help as it would have a conflicting architecture (Mac and *libusb*:`arm64` vs *dfu-util*: `x86_64`).

It's unclear to me how this problem relates to #571 ((*dfu-util fails on Mac M1 when libusb is not x86_64, but native ARM64*).

A proper solution is to fix the library path of the executables (similarly to how *openocd* does it):

**State before**

```
% cd ~/.platformio/packages/tool-dfuutil/bin  
% otool -L dfu-suffix
dfu-suffix:
	/usr/local/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
% otool -L dfu-util  
dfu-util:
	/usr/local/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
```

**Fix**
```
% install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.0.dylib @loader_path/../lib/libusb-1.0.0.dylib dfu-suffix
% install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.0.dylib @loader_path/../lib/libusb-1.0.0.dylib dfu-util
```

**State after**
```
% otool -L dfu-suffix
dfu-suffix:
	@loader_path/../lib/libusb-1.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
% otool -L dfu-util 
dfu-util:
	@loader_path/../lib/libusb-1.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
```

**Full platformio.ini**
```
[env:blackpill-f401cc]
platform = ststm32
framework = libopencm3
board = blackpill_f401cc
upload_protocol = dfu
```

ivankravets, 2022-06-11:
Resolved in https://github.com/platformio/platform-ststm32/pull/630

closed 2022-06-11 by ivankravets

pathnirvana, 2023-02-02:
I am still getting this libusb error on a fresh install of pio on a m1 mac. Even changing the `platform = https://github.com/platformio/platform-ststm32.git` in platformio.ini file does not help.

However I am able to upload manually using the dfu-util that I have install using homebrew

VIPQualityPost, 2023-08-16:
[libusb-1.0.dylib.zip](https://github.com/platformio/platform-ststm32/files/12353901/libusb-1.0.dylib.zip)
I also had this issue. Installing this lib to /usr/local/lib like @Hraph mentioned fixed the problem for me.

bmentink, 2024-04-13:
Hi, is this still the issue? What is being done? I get the following:
```
URRENT: upload_protocol = dfu
Uploading .pio/build/bluepill_f103c8_128k/firmware.bin
dyld[46839]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <196384F2-82B2-3D72-BA1C-183785B95BB9> /Users/bernardmentink/.platformio/packages/tool-dfuutil/bin/dfu-util
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
```

bmentink, 2024-04-13:
I ended up doing it, and I did get further, however I now get the following error:
```
Uploading .pio/build/bluepill_f103c8_128k/firmware.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: No DFU capable USB device available
*** [upload] Error 74
```
The esp32f103 bluepill board does get found by Arduino as a DFU device, also in windows. Just not in platformio om Macos M1

VIPQualityPost, 2024-04-13:
I don't think your error is related to this device, because it is opening dfu-tool (which was not happening in the issue). However you are not finding a dfu device. Are you sure you have a bootloader for blue pill which has USB DFU as a feature? 
You can also try to run `dfu-util --list` from terminal on mac, if there is an eligible DFU device, then it will show up here. But an issue with not showing up is not going to be related to libusb, I think.

bmentink, 2024-04-13:
Hmm, confused then, because as I say the board shows up in Arduino as a DFU device, also in windows, as it want's to install DFU driver. Just does not show up in PIO

I have tried various bootloaders, the last one from [https://github.com/rogerclarkmelbourne/STM32duino-bootloader/blob/master/binaries/generic_boot20_pc13.bin](url)

I have tried installing via st-flash with the st-link v2 dongle, or the uart method with  st32flash.
With stm32flash I flashed to location 0x00, and with st-flash flashed to location 0x8000000
As far as I can tell this is correct?

With one bootloader it flashes on reset, then stops. With another bootloader it flashes fast, then flashes slow ..

I as at a loss in what to try next  ... looks like I am going to have to use st-link for my programming ..

bmentink, 2024-04-13:
I notice that my bluepill has a cheap chinese micro stm32f103c6  not ..c8
I wonder if that is causing issues.

I can't even get a simple cmsis led blink example to run (it uploads fine via st-linkv2 ..

VIPQualityPost, 2024-04-13:
The way that the Arduino DFU upload works is that it tries to open a serial port at 1200 baud on the blue pill, there is some code that notices this and jumps into a special bootloader. There is no DFU port before you do this, and it's a special part of arduino upload that does this - you would have to add that functionality yourself in PIO. PIO is not doing any serial port hack which is why you don't see the DFU port. Perhaps if you set framework as Arduino, then it will try- but definitely it will not work with libopencm3 unless you are adding that yourself.
The roger stuff is ancient and unmaintained I think, and also from well before libopencm3 and ststm32, so you shouldn't use that either...
Unless you want to add DFU yourself, you would save yourself a lot of work to just get a board that supports DFU bootloader natively, like G431KB nucleo (same form factor as blue pill)

bmentink, 2024-04-13:
Thanks, that helps. Did you notice my additional comment. I can't even get a simple blink program to run via st-linkv2, it builds and programs fine.. that was an example direct from platformio ..maybe I have a bad  batch of boards (5. (: )

VIPQualityPost, 2024-04-13:
Sorry, I don't use CMSIS framework, so I don't know how to help. 
This is off topic for the issue, but really there should not be much work in porting code from F1 to G4 unless you were doing things in a really strange way, even if you use ST HAL or Arduino it should be trivial as the hardware not so different. It seems like you are flailing by changing between libopencm3 and CMSIS and wanting to use Arduino bootloaders... if you stick with one thing it will be easier to debug for yourself.

bmentink, 2024-04-14:
No, you don'i understand. My code is written for libopencm3. I was just trying to see if any "blink" program works on the board, so I tryed both cmsis and libopencm3 .... I don't intend to use CMSIS.

I will have a look at porting my libopencm3 existing project to libopencm3 on the G431, however I can't even find any
platformio examples for that target ... unless I am looking in the wrong place.