platformio/platform-ststm32 #596: Board support request: LoRa E5 Dev Board / STM32WLE5JC

nobodywasishere, 2022-02-08:
[This](https://www.seeedstudio.com/LoRa-E5-Dev-Kit-p-4868.html) is the board/mcu I am looking to add support for, which is very similar to #543. 

I have created this board json called `lora_wle5jc.json` that may work, I don't have a board on-hand just yet to test it yet and putting it in `~/.platformio/boards` didn't let it show up in PlatformIO. I don't know of any frameworks that currently support it, but that is okay for what I am hoping to use it for (just bare-metal) (if a framework is required, please let me know). As well, I am not sure what upload protocols it supports just yet.
 
```json
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32WLxx -DSTM32WLE5xx",
    "f_cpu": "48000000L",
    "framework_extra_flags": {
      "arduino": "-DUSE_CM4_STARTUP_FILE"
    },
    "mcu": "stm32wle5jc",
    "product_line": "STM32WLE5xx",
    "variant": "STM32WLxx/WL54JCI_WL55JCI_WLE5JC"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32WLE5JC",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_target": "stm32wlx",
    "svd_path": "STM32WLE5_CM4.svd"
  },
  "frameworks": [
  ],
  "name": "LoRa E5 Dev Board",
  "upload": {
    "maximum_ram_size": 65536,
    "maximum_size": 262144,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "mbed"
    ]
  },
  "url": "https://www.seeedstudio.com/LoRa-E5-Dev-Kit-p-4868.html",
  "vendor": "Seeed"
}
```

The SVD is available [here](https://github.com/fullyautomated/st-svd/blob/main/STM32WLE5_CM4.svd).

nobodywasishere, 2022-02-09:
I have a work-in-progress version going [here](https://github.com/nobodywasishere/platform-ststm32) that I'll update as I test it

nobodywasishere, 2022-02-09:
Also this board is supported by Zephyr and there's useful documentation on how to use it [here](https://docs.zephyrproject.org/2.7.0/boards/arm/lora_e5_dev_board/doc/lora_e5_dev_board.html)

nobodywasishere, 2022-02-11:
I believe i've gotten both the CMSIS and STM32Cube frameworks working, still waiting to get a board to test them but they compile without errors. I'm starting to work on examples for these [here](https://github.com/nobodywasishere/ee193/tree/main/02-blinky). [framework-stm32cubewl](https://registry.platformio.org/tools/out0f0rder/framework-stm32cubewl) is already published to the PIO registry, and I'm currently waiting for [framework-cmsis-stm32wl](https://registry.platformio.org/tools/nobodywasishere/framework-cmsis-stm32wl) to be approved.

nobodywasishere, 2022-02-15:
I've got working blinky programs with both CMSIS and STM32Cube frameworks, and have verified that those work for flashing the board. Should I create a PR with my board file? Currently still waiting for the [framework-cmsis-stm32wl](https://registry.platformio.org/tools/nobodywasishere/framework-cmsis-stm32wl) to be published, so until that happens CMSIS won't work.

drucknase, 2022-05-07:
Hello,
I use platformIO for my LoRa Node development. I bougth Seeed-Studio LoRa-E5 Dev board and want to use it with platformio IDE. Fortunately I found your board request but as far as I found this is until now not included in the platformio repository. So I tried to follow your thread but have no success to get my demo working. What I did:

- open a new platformio project with this platformio.ini file
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:lora_e5_dev_board]
platform = https://github.com/nobodywasishere/platform-ststm32.git
board = lora_e5_dev_board
framework = stm32cube
`

- when I start build I get the following error: 
`Error: Traceback (most recent call last):
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 134, in cli
    process_env(
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 176, in process_env    
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 67, in process       
    result = init_platform(self.options["platform"]).run(
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\platform\_run.py", line 59, in run
    self.configure_default_packages(options, targets)
  File "C:\Users\wolle\.platformio\platforms\ststm32@src-82efdcce516976d299fceeabed7df5a7\platform.py", line 110, in configure_default_packages
    return PlatformBase.configure_default_packages(self, variables,
  File "C:\Users\wolle\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 194, in configure_default_packages
    self.packages[_pkg_name]["optional"] = False
KeyError: 'framework-stm32cubewl'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================`

- what went wrong?
- where can I find your blinky example?
- normaly I use arduino framework. Is it possible to extend your work also for arduino framework?

Thank's
Wolfgang

nobodywasishere, 2022-05-07:
What's happening is the proper framework package isn't being found. I believe you either need to add

```

```

nobodywasishere, 2022-05-07:
Unfortunately i could not get the arduino framework working at this time, so your options are cmsis, stm32cube, and zephyr

drucknase, 2022-05-07:
thank you for your fast response. I will try your proposal.

PR #624 (2023-09-19): Add support for LoRa E5 Dev Board
Closes #596. Can be squashed.
closed 2023-09-19 by valeros