platformio/platform-ststm32 #602: Update Zephyr to v3.0.0

nobodywasishere, 2022-02-24:
New release is available: https://github.com/zephyrproject-rtos/zephyr/releases/tag/zephyr-v3.0.0

AlexBollmann, 2022-04-22:
Is this planned to be updatd? would be very nice to have version 3 available

lsangild, 2022-04-27:
Before this happens, it would be nice to have an intermediary with [Zephyr 2.7.2](https://github.com/zephyrproject-rtos/zephyr/releases/tag/v2.7.2).
Zephyr 2.7 is an LTS, so it would be nice to have access to the current LTS (for production) alongside the option to switch to the newest release for testing.
Maybe rename the Zephyr LTS framework to zephyr_lts or something, so both branches are supported.

Hygdrasil, 2022-12-23:
Im working on a integration of Zephyr 3 based on   [maxgerhardts](https://github.com/maxgerhardt/zephyr/tree/3.1.0_pio) work.
 See [https://github.com/Hygdrasil/platformIO_zephyr3/tree/piosupport31](https://github.com/Hygdrasil/platformIO_zephyr3/tree/piosupport31).
 Any help is appreciated.

lsangild, 2023-03-22:
Zephyr already has a fine build system called `west`. Is there a reason that PlatformIO does not just become a wrapper around west commands? And in addition use the official Zephyr SDK.
Would that not be easier than the dependency on [platformio/toolchain-gccarmnoneeabi](https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi).

Is that where you are trying to do @Hygdrasil and @maxgerhardt ?

This way it would be possible to set Zephyr versions rather easily with the [west init](https://docs.zephyrproject.org/latest/develop/west/basics.html#west-init-and-west-update) command.

I am willing to attempt this adventure, but I have a hard time wrapping my head around the PlatformIO package registry, so I need to know which repos need updating to allow me to get a working version of Zephyr by specifying e.g. `platform = ststm32@16.x.x` in my `platform.ini` file.

I guess a package(tool) called `west` would have to be added to the repo, and the bindings for west to PlatformIO commands should be added. Additionally the `platformio/framework-zephyr` would have to depend on this west package, but how can I update that package?

This way, whenever there is a need to update Zephyr, it would optimally be possible to just update `platformio/framework-zephyr` with a new tag from the Zephyr repo, and possibly an update of the SDK version, then West handles the rest.

Hygdrasil, 2023-03-22:
Its not the problem to get all the files on the pc.
Its to know that file is needed for the build and that to autogenerate.
In the step between v2.7 and v3 the buildprocess has changed. So we need to do other things with the files from the zephyr repo to generate a valid firmware.

Every new versions can have changes on the build-system (west). These changes have to be pathed to the  other buildsystem (platformIO). Sadly platformIO(SCons) and west (cmake) are not compatible with each other. So there is no easy path.

A nother aproach would be to call west build from platformIO but I havend found an option to let pio output the files it wants to compile. 

I think it build the approach to download all the zephyr dependencies already. Its not quit platformIO like. But could be a starting point for you. I can send it if i havend deleted the test

valeros, 2025-07-23:
Resolved in the latest versions.

closed 2025-07-23 by valeros