platformio/platform-ststm32 #608: Could not initialize project, Python 3.10+

imsakg, 2022-03-21:


zayats1, 2022-03-29:
same problem

imsakg, 2022-04-03:
@zayats1
Hey, i fixed my problem with using platformio core. I also patch some package files because im using Python3.10+.
Changes:
+ /home/$USER/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/types/oldstr.py
Line 5 :
+  from collections import Iterable >  from collections.abc import Iterable 

+ /home/$USER/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/types/oldstr.py
Line 4:
+  from collections import Mapping > from collections.abc import Mapping

zayats1, 2022-04-07:
@imsakg Thank you! It works.

matteomeneghetti, 2022-07-05:
I have encountered this issue as well and the easy fix provided by @imsakg worked fine.
I suppose the python3 dependencies list should be updated to include the most recent _python-future_ package version (which contains the _past_ package as well), as shown for example [here](https://github.com/PythonCharmers/python-future/blob/master/src/past/types/oldstr.py) and [here](https://github.com/PythonCharmers/python-future/blob/master/src/past/builtins/misc.py)?

mattj23, 2022-09-09:
Python 3.10 has been out for almost a year now, it's probably worth trying to get these fixes into the main release.  Is this the right repository for a fix?

qleonardolp, 2023-03-07:
> @zayats1 Hey, i fixed my problem with using platformio core. I also patch some package files because im using Python3.10+. Changes: ~ /home/$USER/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/types/oldstr.py Line 5 :
> 
>     * from collections import Iterable >  from collections.abc import Iterable
> 
> 
> ~ /home/$USER/.platformio/packages/framework-mbed/platformio/package_deps/py3/past/builtins/misc.py Line 4:
> 
>     * from collections import Mapping > from collections.abc import Mapping

After these steps is it necessary any other configuration? I edited the files, however the problem persists

imsakg, 2023-03-07:
> After these steps is it necessary any other configuration? I edited the files, however the problem persists

Hey, you can trace your stderr message and find the problem. It was enough to edit that file before but idk now.

closed 2023-03-07 by imsakg