platformio/platform-ststm32 #610: Installed version of STM32Flash (v. 0.4) does not support newer chips (e.g. STM32F4)

talikarng, 2022-03-29:
Fresh install of the STM32 platform on windows uses STM32Flash version 0.4 which does not support newer chips (e.g. STM32F4).
When attempting to upload to the STM32F4, I would get an error message along the lines of ("unknown/unsupported device 0x431") - I can't recall the exact error message - and would fail to upload any sketch.
Checking the STM32Flash gitlab page reveals that device 0x431 (STM32F4...) was included in version 0.5 onwards.

Manually installing a newer version of STM32Flash.exe into ......platformio\packages\tool-stm32duino resulted in being able to upload to the STM32F4 board.

dalbert2, 2022-07-13:
+1

stm32flash.exe version 0.4 is badly out of date and doesn't support many newer ST processors (e.g. STM32L471). Replacing with the latest stm32flash.exe version (0.7) fixes this.

Latest version is available here: https://sourceforge.net/projects/stm32flash/
Replacement goes in $USER/.platformio/packages/tool-stm32duino/stm32flash

travisjayday, 2025-04-03:
Version 0.4 did not work with my STM32L151C8 target. Some cryptic message about failed to erase chip. Manually replacing the file with v0.7 made it work. Would be great if this could be updated somehow.

commit 4b5ff01c (2025-04-07): Switch to a standalone stm32flash package

The new package supersedes the outdated tool-stm32duino package

Resolves #610, resolves #640, resolves #665
closed 2025-04-07 by valeros