platformio/platform-ststm32 #616: Can't print floating point values by default on nucleo_f446r

AaronErhardt, 2022-04-10:
If I want to write code like this:

```c
float num = 1.0;
printf("num = %f\n", num);
```

I need this to be able to print floating point values:

```c
asm(".global _printf_float");
```

I think this should be enabled by default and not require a this hack to get it working.

### My configuration

```ini
[env:nucleo_f446re]
platform = ststm32
board = nucleo_f446re
framework = stm32cube

build_flags = 
    -D USE_FULL_LL_DRIVER
```

maxgerhardt, 2022-06-13:
>I think this should be enabled by default 

Then firmware size blows up [really bad.](https://community.platformio.org/t/firmware-size-increased-more-than-twice-after-adding-printf-float-flag/15539)

AaronErhardt, 2022-06-13:
> Then firmware size blows up really bad.

I didn't know about that. Makes sense why this isn't enabled by default now.

Yet, I wish the compiler would be smart enough to throw a warning when you try to print floats while this feature is disabled.

closed 2022-06-13 by AaronErhardt