platformio/platform-ststm32 #628: genericSTM32F103Vx Clock Frequency

nopnop2002, 2022-05-25:


maxgerhardt, 2022-06-13:
>genericSTM32F103VB/VC/VD/VE
>
>The clock frequency is set to 64MHz instead of 72MHz.

Board maps to variant

https://github.com/platformio/platform-ststm32/blob/9ce784b031f5fb0dfac6dd5cda33ef4ab58191db/boards/genericSTM32F103VB.json#L19

And https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32F1xx/F103V8(H-T)_F103VB(H-I-T) only has one clock code in [generic_clock.c](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103V8(H-T)_F103VB(H-I-T)/generic_clock.c). If the USB communication is enabled, it clocks the CPU by HSI / 2 * 12, with HSI = 8MHz, that is 48 Mhz, else, by HSI / 2 * 16 = 64MHz.

If you have a high-speed crystal (HSE) on the board, you can achieve the full frequency, which is what PlatformIO always lists. It does not account for default clock init code of different frameworks. To do this, you have to place a function in your firmware, e.g. in the `src/main.cpp` file, with

```cpp
extern "C" void SystemClock_Config(void)
{
  // clock init code here
}
```
and that inner code can e.g. be generated with STM32CubeMX's clock configurator tool.

nopnop2002, 2022-06-13:
I did another test with a bluepill board

# blackpill_f103c8             STM32F103C8T6   72MHz        64KB      20KB      BlackPill F103C8

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = bluepill_f103c8

[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
```

F_CPU show 72000000

# genericSTM32F103C8           STM32F103C8T6   72MHz        64KB      20KB      STM32F103C8 (20k RAM. 64k Flash)

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = genericSTM32F103CB

[env:genericSTM32F103CB]
platform = ststm32
board = genericSTM32F103CB
framework = arduino
```

F_CPU show 64000000

maxgerhardt, 2022-06-13:
`board = bluepill_f103c8` shows F_CPU = 72000000 because it [its clock code configures it to do so](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103C8T_F103CB(T-U)/variant_PILL_F103Cx.cpp#L103-L108), HSE (=8MHz) / 1 * 9 = 72 MHz.

`board = genericSTM32F103C8` shows  F_CPU = 64000000 because it [its clock code configures it to do so](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32F1xx/F103C8T_F103CB(T-U)/generic_clock.c#L68-L72), HSI (=8MHz) / 2 * 16 = 64 MHz.

nopnop2002, 2022-06-13:
@maxgerhardt 

Thank you for the commentary.

I would like to keep this issues open for others.

nopnop2002, 2022-06-13:
@maxgerhardt 

Don't you think 72Mhz on this list should be changed?

[image]

nopnop2002, 2022-06-13:
@maxgerhardt 

I got it.

valeros, 2023-06-14:
The board manifest contains maximum chip frequency that has nothing to do with specific particular settings.

closed 2023-06-14 by valeros